VLSIDA/OpenRAM #149: Installation issue

Ryabkomv, 2022-07-22:
I tried OpenRAM and after [comment ](https://github.com/VLSIDA/OpenRAM/issues/146#issuecomment-1191704852)decided to make clean installation.

My steps are:

```
sudo git clone https://github.com/VLSIDA/OpenRAM.git
cd OpenRAM//docker/
sudo make build
```

I got an error:

```
Step 7/74 : RUN apt-get install --no-install-recommends -y build-essential git ssh vim gosu autoconf automake libtool bison flex
 ---> Running in d0575047808a
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package build-essential
E: Unable to locate package git
E: Unable to locate package ssh
E: Unable to locate package vim
E: Unable to locate package gosu
E: Unable to locate package autoconf
E: Unable to locate package automake
E: Unable to locate package libtool
E: Unable to locate package bison
E: Unable to locate package flex
The command '/bin/sh -c apt-get install --no-install-recommends -y build-essential git ssh vim gosu autoconf automake libtool bison flex' returned a non-zero code: 100
docker tag  vlsida/openram-ubuntu:2022-07-22 vlsida/openram-ubuntu:latest
Error response from daemon: No such image: vlsida/openram-ubuntu:2022-07-22
make: *** [Makefile:11: build] Error 1
```

Is the error related to docker file, or this is my system issue? By the way, running in terminal 
`apt-get install --no-install-recommends -y build-essential git ssh vim gosu autoconf automake libtool bison flex`
works without any issue.

googling suggested to install `apt-transport-https` but it already installed.

mguthaus, 2022-07-22:
This looks like a docker permissions problem because you built with sudo. I'd suggest deleting the image and building without sudo. You can add yourself to the docket group so that sudo isn't necessary.

closed 2022-07-22 by mguthaus

Ryabkomv, 2022-07-22:
Unfortunately it doesn't help. I follow the [steps](https://docs.docker.com/engine/install/linux-postinstall/) to run docker without sudo, now it runs `docker hello-world` successfully. 

However with OpenRAM I still got the same error.

By the way, I installed all the tools in docker image for OpenLane with sudo successfully. 

I appreciate if you can further help to resolve the issue.

mguthaus, 2022-07-22:
What type of system is the host? Looking closer it could be a network problem too. I'm not an expert in docker so I'd suggest getting help through their forums.

Ryabkomv, 2022-07-22:
I'm using Ubuntu 20.04.4 LTS
Ok, thank you. I just want to confirm that I'm doing the things right.
1. I just clone the repository, go to docker folder, and run `make build` right?

mguthaus, 2022-07-22:
I build on the same all the time so I'm not sure what is going on...

mguthaus, 2022-07-22:
Yes, those instructions are right.

mguthaus, 2022-07-22:
What version of docker are you using?

Ryabkomv, 2022-07-22:
Docker version 20.10.17, build 100c701

mguthaus, 2022-07-22:
I'm using the exact version. I can't replicate your problem....

Ryabkomv, 2022-07-27:
Looks like bad luck. I still have no idea what is going on.
Is there a way to install OpenRAM manually?

mguthaus, 2022-07-27:
You can look through the versions of tools required in the Dockerfile and install those locally but this isn't recommended.

I have a number of people using the same configuration so I'm not sure what else it could be. If you can isolate it a bit more, maybe we can help. 

Does it recur every time? Sometimes network problems are intermittent.

Can you manually apt install in a docker image?

Ryabkomv, 2022-07-28:
I tried to install on virtual Ubuntu machine which uses different network. Process is still ongoing but passed the stage at which the error is occurred previously. However I notice some other errors during installation which doesn't interrupt the process. Hopefully it is not fatal.

Meantime, I have not found in documentation write mask support. But some student [work ](https://escholarship.org/uc/item/2476f8tv)  mentioned about such functionality. 

My interest to OpenRAM is to combine Tiny Rocket core with photonic device and pass it through the OpenLane flow to get GDSII. But Rocket core uses write mask functionality which might be not yet supported by OpenRAM?

mguthaus, 2022-07-28:
What errors? Errors are usually a problem whereas warnings may not be.

The write_size option determines the write granularity and hence the write size. It is fully supported. You should join the slack for further usage questions.

Ryabkomv, 2022-07-28:
Finally I got the following error:
```
make[3]: Leaving directory '/root/klayout/build-debug/gsiqt/qt5/QtMultimedia'
make[2]: Leaving directory '/root/klayout/build-debug/gsiqt/qt5'
make[1]: Leaving directory '/root/klayout/build-debug/gsiqt'
The command '/bin/sh -c ./build.sh -qt5 -debug -j 8  && cp -r bin-debug /usr/local/klayout' returned a non-zero code: 1
docker tag  vlsida/openram-ubuntu:2022-07-28 vlsida/openram-ubuntu:latest
Error response from daemon: No such image: vlsida/openram-ubuntu:2022-07-28
make: *** [Makefile:11: build] Error 1
```

mguthaus, 2022-07-28:
I don't know what you ran to get that... Can you share the entire output?

mguthaus, 2022-07-28:
Actually it looks like you just ran make build. It seems like you have a system problem if klayout is failing to build. It builds fine here... What is the error from it?

mguthaus, 2022-07-28:
Oh, can you decrease the number of threads given to the klayout build? Right now it is trying to use 8. Do you have 8 cores?

Ryabkomv, 2022-07-29:
I've noticed that virtual machine by default uses 1 core. So I changed the line in Dockerfile:
```
RUN ./build.sh -qt5 -debug -j 1 \
 && cp -r bin-debug /usr/local/klayout
```
to use single core, but still have the same error:

```
../../../src/rdb/rdb/gsiDeclRdb.cc: In function 'void gsi::tp_output_rdb(db::TilingProcessor*, const string&, rdb::Database&, rdb::id_type, rdb::id_type)':
../../../src/rdb/rdb/gsiDeclRdb.cc:1388:30: error: expected type-specifier
 1388 |   proc->output (name, 0, new rdb::TiledRdbOutputReceiver (&rdb, cell_id, category_id), db::ICplxTrans ());
      |                              ^~~
../../../src/rdb/rdb/gsiDeclRdb.cc:1386:95: warning: unused parameter 'rdb' [-Wunused-parameter]
 1386 | static void tp_output_rdb (db::TilingProcessor *proc, const std::string &name, rdb::Database &rdb, rdb::id_type cell_id, rdb::id_type category_id)
      |                                                                                ~~~~~~~~~~~~~~~^~~
../../../src/rdb/rdb/gsiDeclRdb.cc:1386:113: warning: unused parameter 'cell_id' [-Wunused-parameter]
 1386 | static void tp_output_rdb (db::TilingProcessor *proc, const std::string &name, rdb::Database &rdb, rdb::id_type cell_id, rdb::id_type category_id)
      |                                                                                                    ~~~~~~~~~~~~~^~~~~~~
../../../src/rdb/rdb/gsiDeclRdb.cc:1386:135: warning: unused parameter 'category_id' [-Wunused-parameter]
 1386 | static void tp_output_rdb (db::TilingProcessor *proc, const std::string &name, rdb::Database &rdb, rdb::id_type cell_id, rdb::id_type category_id)
      |                                                                                                                          ~~~~~~~~~~~~~^~~~~~~~~~~
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-reserved-user-defined-literal'
make[2]: *** [Makefile:561: gsiDeclRdb.o] Error 1
make[2]: Leaving directory '/root/klayout/build-debug/rdb/rdb'
make[1]: *** [Makefile:48: sub-rdb-make_first] Error 2
make[1]: Leaving directory '/root/klayout/build-debug/rdb'
make: *** [Makefile:143: sub-rdb-make_first] Error 2
The command '/bin/sh -c ./build.sh -qt5 -debug -j 1  && cp -r bin-debug /usr/local/klayout' returned a non-zero code: 1
docker tag  vlsida/openram-ubuntu:2022-07-29 vlsida/openram-ubuntu:latest
Error response from daemon: No such image: vlsida/openram-ubuntu:2022-07-29
make: *** [Makefile:11: build] Error 1
```

I wonder why it says 
`No such image: vlsida/openram-ubuntu:2022-07-29`
same as it was with the first machine.

mguthaus, 2022-07-29:
You get that message because it doesn't complete building the docker image. The build command tries to create an image with today's date and if it succeeds, it tags it as the latest. I don't do any error checking if the build actually completes...

Ryabkomv, 2022-07-31:
Ok. I have reinstalled everything from scratch. I set 4 cores on my virtual machine and started installation.
I get error at the same step.
```
mv -f libklayout_QtMultimedia.so.0.27 ../../../../build-debug/libklayout_QtMultimedia.so.0.27
make[3]: Leaving directory '/root/klayout/build-debug/gsiqt/qt5/QtMultimedia'
make[2]: Leaving directory '/root/klayout/build-debug/gsiqt/qt5'
make[1]: Leaving directory '/root/klayout/build-debug/gsiqt'
The command '/bin/sh -c ./build.sh -qt5 -debug -j 4  && cp -r bin-debug /usr/local/klayout' returned a non-zero code: 1
make: *** [Makefile:10: build] Error 1
```
I have no idea why this is happening. I had issue on 2 different machines. If nobody experienced such issue before seems my case is very specific. 

And I do not think klayout itself is an issue, because I have installed it successfully both as a separate installation and in the scope of docker image with OpenLane.

mguthaus, 2022-07-31:
Can you find more information on why klayout failed to build? The return code alone isn't sufficient to understand why and fix it. 

I've also installed this on at least 3 machines, my students on all of their machines, and none of them did this.

Ryabkomv, 2022-07-31:
I'm not and expert to be able to fix the build flow.
If you can suggest how to attack the problem I will check.

mguthaus, 2022-07-31:
Can you include the whole output as an attachment?

Ryabkomv, 2022-07-31:
ok, thanks. I have started the installation with redirection of the output to the file. Once finished I will share the file.

Ryabkomv, 2022-08-01:
Here is a [file](https://github.com/Ryabkomv/openroad_issue_reproducible/blob/master/my.txt)

mguthaus, 2022-08-01:
It looks like you might have some permission problems:

> docker build -t vlsida/openram-ubuntu:2022-07-31 -f Dockerfile . | tee -i openram-ubuntu.log                                                                               
> tee: openram-ubuntu.log: Permission denied 

Do you not have permissions on your local directory?

Also, do you not have permissions to write /usr/local? Maybe that is the problem...

mguthaus, 2022-08-01:
Can you change this line to two lines?

```
RUN ./build.sh -qt5 -debug -j 8 \
   && cp -r bin-debug /usr/local/klayout
```
   
   like this:
   
```
  RUN ./build.sh -qt5 -debug -j 8
  RUN cp -r bin-debug /usr/local/klayout
```
Does it fail on the second line? It seems to be completing the build in your output.

mguthaus, 2022-08-01:
Can you send me the stdout in openram-ubuntu.log when you figure out the permissions problem?

Ryabkomv, 2022-08-02:
With suggested split:
```
RUN ./build.sh -qt5 -debug -j 8
RUN cp -r bin-debug /usr/local/klayout
```

installation has passed this stage, but failed to install `ngspice ` due to failure in name resolution of `git://git.code.sf.net/p/ngspice/ngspice `. Is it the Issue from my side only? 
I have changed the repository to `https://github.com/ngspice/ngspice`

with that the installation has finished.

I was also able to compile the simple memory example from the main page.

Will check further with sky130 pdk

Ryabkomv, 2022-08-02:
> Splitting the commands wasn't a suggested fix, it was supposed to help me
> understand what was the error. Whether it was a build problem or a copy
> problem. The fact that it passes with that makes me believe there is
> something very unstable in your system or setup.
>
> What about the initial permissions issue? Did you fix that? Why don't you
> have permissions to write the local directory?
>
> The network problem is due to an unstable network. Try again and/or try on
> a better network.

I'm not sure if permission issues are still there. The mentioned folders
are owned by root. So I made installation with sudo.

Network during installation was different, but I think both of networks
should have similar restrictions.

I also wonder why the split of the commands allow to overcome the issue. I
remember some suggestions at stackoverflow to split commands in docker file.

Anyway, I will check if I can compile needed memory using sky130.

mguthaus, 2022-08-02:
That ngspice repo hasn't been updated since 2015. It may not work...

Ryabkomv, 2022-08-02:
Oh, but I can't access original repo from any network. Will try to reinstall with up to date repo later.
As of now I was able to compile SRAM with default library `scn4m_subm` and 
```
# Data word size
word_size = 32
# Number of words in the memory
num_words = 1024
```
However when I change library to `tech_name = "sky130"`

I got the following error:
```
** Start: 08/02/2022 17:29:46
Technology: sky130
Total size: 32768 bits
Word size: 32
Words: 1024
Banks: 1
Write size: None
RW ports: 1
R-only ports: 0
W-only ports: 0
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
Only generating nominal corner timing.
ERROR: file sram_config.py: line 132: Invalid number of cols including rbl(s): 257. Total cols must be divisible by 2
Traceback (most recent call last):
  File "/home/maxim/OpenRAM/compiler/openram.py", line 54, in <module>
    c = sram_config(word_size=OPTS.word_size,
  File "/home/maxim/OpenRAM/compiler/modules/sram_config.py", line 44, in __init__
    self.compute_sizes()
  File "/home/maxim/OpenRAM/compiler/modules/sram_config.py", line 95, in compute_sizes
    self.recompute_sizes()
  File "/home/maxim/OpenRAM/compiler/modules/sram_config.py", line 132, in recompute_sizes
    debug.error("Invalid number of cols including rbl(s): {}. Total cols must be divisible by {}".format(self.num_cols + num_ports + self.num_spare_cols, self.array_col_multiple), -1)
  File "/home/maxim/OpenRAM/compiler/debug.py", line 47, in error
    assert return_value == 0
AssertionError

```
I might did something stupid, because I just changed the pdk, although config is the same to [this](https://github.com/ShonTaware/SRAM_SKY130)

my full config is 
```
# Data word size
word_size = 32
# Number of words in the memory
num_words = 1024

# Technology to use in $OPENRAM_TECH
tech_name = "sky130"

# You can use the technology nominal corner only
nominal_corner_only = True
# Or you can specify particular corners
# Process corners to characterize
# process_corners = ["SS", "TT", "FF"]
# Voltage corners to characterize
# supply_voltages = [ 3.0, 3.3, 3.5 ]
# Temperature corners to characterize
# temperatures = [ 0, 25 100]

# Output directory for the results
output_path = "temp"
# Output file base name
output_name = "sram_{0}_{1}_{2}".format(word_size,num_words,tech_name)

# Disable analytical models for full characterization (WARNING: slow!)
# analytical_delay = False
```

mguthaus, 2022-08-02:
Take a look at the example configs in:
https://github.com/VLSIDA/OpenRAM/blob/0d616ae072e6c42a0d8a006eebc681408502e956/macros/configs/sky130_sram_1kbyte_1rw1r_32x256_8.py

sky130 has certain requirements on the number of columns and rows. Since we use a replica timing column, there must be a dummy column added to make the number even.

mguthaus, 2022-08-02:
I'm not sure what to say about the ngspice repo. Their official distribution is via sourceforge (sf.net)... Can you access this information page?
https://ngspice.sourceforge.io/gitaccess.html

Ryabkomv, 2022-08-02:
yes, this page is accessible!

mguthaus, 2022-08-02:
But it says where to get the git repo which is unaccessible to you. Maybe you should contact them?

Ryabkomv, 2022-08-02:
Thanks, I will try. Actually, when I first tried manual installation of the
OpenLane, some of the repos were at  sourceforge, and were inaccessible to
me as well. I saw some people also experienced issues with  sourceforge.
I will contact them.