VLSIDA/OpenRAM #152: Test fails when unit testing

jingyao-zhang, 2022-07-25:
Hi everyone. I have a problem when I am running unit test `21_ngspice_delay_test`. The tech `scn4m_subm` can pass the test but `freepdk45` cannot. According to the `sram.log` file, the error is
`ERROR: file delay.py: line 708: Timed out, could not find a feasible period.`
Is there any way I can get the relative accurate delay under `freepdk45` ?

mguthaus, 2022-07-25:
This is an issue that the PTM models from FreePDK45 aren't included by default. It can't find the models and so simulations fail with whatever default device models are in ngspice.

jingyao-zhang, 2022-07-26:
Thanks for your help!
I have successfully passed that unit test under `freepdk45`.

mguthaus, 2026-10-13:
Closing as stale. This has been inactive for 4.2 years and there is no current reproduction against `dev`, so it is not actionable as it stands.

The unit test harness has changed since this was filed; `compiler/tests/Makefile` is the current entry point and tracks known-failing cases in `BROKEN_STAMPS`.

To be clear, this is being closed for inactivity rather than rejected. If it still affects you, please reopen or file a fresh issue with a reproduction against current `dev`, including your configuration file, technology, OS and `python3 --version`, and the contents of the temp directory OpenRAM leaves behind on failure ([the FAQ explains which files those are](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself, a PR would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch.

closed 2026-10-13 by mguthaus