VLSIDA/OpenRAM #155: README.md: broken link to freepdk45

vmiheer, 2022-09-02:
> You may get the entire [FreePDK45 PDK here](https://www.eda.ncsu.edu/wiki/FreePDK45:Contents).

It would be good to have this link fixed.

mguthaus, 2022-09-02:
I've reported the link as broken last week. Unfortunately, I don't have an updated one. I'll leave this open for now until they fix it.

vmiheer, 2022-09-03:
> Note, it was also behind a registration wall. I'm trying to convince them
> to just add it to GitHub.

Thank you, Matt!

ashkanr65, 2022-11-22:
Hey @mguthaus ,
I get this error when I run "make pdk":

`Cloning skywater PDK...
Previous HEAD position was f70d8ca Merge pull request #346 from QuantamHD/add_reram
Switched to branch 'main'
Your branch is up to date with 'origin/main'.
Already up to date.
Cloning open_pdks...
You are not currently on a branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>
`
How can I solve this issue?

mguthaus, 2022-11-22:
Which commit are you using?

What if you do "make wipe" first to erase the old versions?

mguthaus, 2022-11-22:
BTW, this is a new issue so re-opening this one doesn't make sense...

mguthaus, 2026-10-13:
Still open and about as small as a fix gets. The README points `[FreePDK45]` at `https://www.eda.ncsu.edu/wiki/FreePDK45:Contents`. If that is still dead, a one-line PR updating the link — in `README.md` and in `docs/source/basic_setup.md`, which carries the same reference — would close this out.

If you would like to take it, a PR against `dev` would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch, and the regression suite is run with `cd compiler/tests && make -k -j<N>`.

mguthaus, 2026-10-20:
Fixed on `dev`. #330 (merged as f2ac032a) updated the `[FreePDK45]` link in `README.md` and `docs/source/basic_setup.md`, and I pushed d78e82a9 for the two places it missed.

Verified both ends:

| URL | result |
|---|---|
| `https://www.eda.ncsu.edu/wiki/FreePDK45:Contents` | redirects to `eda.ncsu.edu`, then **404** |
| `https://eda.ncsu.edu/freepdk/freepdk45/` | **200**, title "FreePDK45 \| NC State EDA", the real resource page |

The dead URL turned out to be in four files, not two:

- `README.md` and `docs/source/basic_setup.md` — fixed by #330
- `PORTING.md` — the same `[FreePDK45]:` reference definition, fixed in d78e82a9
- `technology/freepdk45/tech/freepdk45.lydrc` — three URLs in comments, including the `FreePDK45:RuleDevel` and `FreePDK45:Manufacturing_Grid` sub-pages, which are also 404 and have no direct replacement, so they now point at the same landing page

There is no longer any reference to the retired NC State wiki anywhere in the tree. Thanks for reporting it, and sorry it sat for four years.

closed 2026-10-20 by mguthaus