VLSIDA/OpenRAM #156: How Can I remove supply pins in Sky130

whutddk, 2022-09-09:
I got two supply ports in my generated SRAM like this. 
 ```
`ifdef USE_POWER_PINS
    vccd1,
    vssd1,
`endif
```

When I running MPW flow, I got 8 nets missing in LVS, when I used 4 SRAM macro.

```
Circuit 1 contains 131554 devices, Circuit 2 contains 131554 devices.
Circuit 1 contains 108965 nets,    Circuit 2 contains 108957 nets. *** MISMATCH ***
```

Question 1:
Is it necessary to add the supply ports and connections to each hierarchy?

Question 2:
Is there any way to remove supply ports in SRAM's interface? My Code is generated from Chisel3, and I have to add connections every time I generated the new code.

mguthaus, 2022-09-09:
These are needed so pdngen knows what to connect... I'm not sure about your use case.

mguthaus, 2022-09-09:
I would look at the LVS report. It's likely something else is disconnected. If there is an issue with OpenRAM, let me know and we can reopen this.

closed 2022-09-09 by mguthaus

whutddk, 2022-09-09:
I upload the lvs log here
[39lef.zip](https://github.com/VLSIDA/OpenRAM/files/9532152/39lef.zip)

mguthaus, 2022-09-09:
It looks like there's more than just 8 nets... take a look at this fragment for example:

626559 Instance: sky130_fd_sc_hd__nand4b_1:i_Rift |Instance: \i_Rift2LinkA.i_rift2Core.if4.in
 626560   A_N = 6                                  |  A_N = 6
 626561   B = 6                                    |  B = 6
 626562   C = 6                                    |  C = 6
 626563   D = 6                                    |  D = 6
 626564   VGND = 329689                            |  VGND = 329699
 626565   VNB = 329689                             |  VNB = 329699
 626566   VPB = 329689                             |  VPB = 329699
 626567   VPWR = 329689                            |  VPWR = 329699
 626568   Y = 4                                    |  Y = 4
 6
 
 That's off by in the VGND and VPWR. It looks like your power routing was incomplete maybe? Sorry, I can't really help more.