VLSIDA/OpenRAM #157: Internal power is super large

Jian-PENG207, 2022-09-13:
Hi，

I use sky130 to generate the SRAM. But in the .lib file, the internal power is super large which is impossible in the real world. This super large internal power result is also reported in the DC. Is this a bug? 
        internal_power(){
            when : "!csb0 & web0"; 
            rise_power(scalar){
                values("1.706915e+11");
            }
            fall_power(scalar){
                values("1.706915e+11");
            }
        }

mguthaus, 2022-09-13:
Yes 

Are you running in analytical mode?

Jian-PENG207, 2022-09-13:
> Yes
> 
> Are you running in analytical mode?

Yes. I would like to get a reliable power simulation based on the .saif file. Is there any other method to get the accurate power consumption with the generated sram model?

mguthaus, 2022-09-13:
This capability doesn't exist. We use an analytical model based on simple
switching assumptions much like CACTI.

closed 2022-09-14 by Jian-PENG207