VLSIDA/OpenRAM #161: How can I implement a 2rw-dual port memory?

highsharp-wyj, 2022-10-27:
hi all,
In scn4m_subm tech,  i want to implement a 2rw-dual port memory, which means,
num_rw_ports = 2 
num_r_ports = 0
num_w_ports = 0
but it shows errors as, 
ERROR: file delay.py: line 882: Sense amp enable timing error. Increase the delay chain through the configuration file.
Could you please help check? Thank you.

mguthaus, 2022-10-27:
What is the rest of your config? This is not an error with the 2 rw ports but a timing control problem.

mguthaus, 2022-10-27:
Also, what commit id are you using?

highsharp-wyj, 2022-10-27:
Thank you. my config file like this,
# Data word size
word_size = 2
# Number of words in the memory
num_words = 16

# Technology to use in $OPENRAM_TECH
tech_name = "scn4m_subm"

# You can use the technology nominal corner only
nominal_corner_only = True   

# Output directory for the results
output_path = "temp"
# Output file base name
output_name = "sram_{0}_{1}_{2}".format(word_size,num_words,tech_name)

# Disable analytical models for full characterization (WARNING: slow!)
analytical_delay = False

highsharp-wyj, 2022-10-27:
> 

i have done this in both stable and latest dev branch, it is the same issue.
commit 0d616ae072e6c42a0d8a006eebc681408502e956
commit 9b6eb4a120f0f85b568da77c641e997a57dc556b

highsharp-wyj, 2022-10-27:
one more question, there are  repetitive information like this in the latest dev branch, when i run openram.py, but it seems this does not affect the result. 

ERROR: file bitcell_base.py: line 274: Must override build_graph function in bitcell base class.

mguthaus, 2022-10-27:
This is a tiny, tiny SRAM. I'm not surprised that it doesn't work. I'd suggest using flip-flops (it's only 32 of them!) or increasing the SRAM size.

closed 2022-10-27 by mguthaus