VLSIDA/OpenRAM #163: Create yosys memlib mapping

xobs, 2022-11-25:
**Describe the bug**
Yosys supports the creation of memory libmap files in order to infer RAMs from chunks of registers. For example, this cache infers BRAMs on most FPGAs:

```
  (* ram_style = "block" *) reg [7:0] ways_3_data_symbol3 [0:1023];
```

Currently, there appears to be no `memory_libmap` pass for OpenRAM, so this gets turned into DFFs instead.

**Version**
c2333394e0b0b9d9d71185678a8d8087715d5e3b from https://github.com/VLSIDA/sky130_sram_macros

**To Reproduce**
Create a register in Verilog annotated with `(* ram_style = "block" *)`.

**Expected behavior**
Yosys should infer one or more RAMs.

**Additional context**
The memory lib pass is documented at https://github.com/YosysHQ/yosys/blob/982a11c709b4b363f85ae52a127f8a98bda30a3f/passes/memory/memlib.md and the required files (techmap and libmap) seem like they should be able to be generated programmatically during the build.

The libmap file seems like it should be easiest, but the techmap Verilog file also seems fairly trivial for someone who is very familiar with Verilog and RAM terminology.

mguthaus, 2022-11-25:
Thank you for documenting. This has been on our radar for a while but we need to make it happen...

xobs, 2022-11-25:
Note that this appears to be slightly more complex due to oddly-shaped memories. For example, this VexRiscv has a cache tag that's 22 bits and Yosys is unable to figure out how to cram that into an 8-bit meory:

```
(* ram_style = "block" *) reg [21:0] ways_1_tags [0:127];
```

This produces:

```
3.2. Analyzing design hierarchy..
Top module:  \VexRiscv
Used module:     \DataCache
Used module:     \InstructionCache
Removed 0 unused modules.

4. Executing MEMORY_LIBMAP pass (mapping memories to cells).
ERROR: no valid mapping found for memory InstructionCache.ways_1_tags
child process exited abnormally
```

mguthaus, 2026-10-13:
Still open and nobody has started on it. A yosys `memlib` mapping is reasonably self-contained: it is a description of the generated macro's ports and timing rather than a change to the generator, so it does not need to touch the compiler internals. The Liberty and Verilog writers in `compiler/characterizer/lib.py` and `compiler/base/verilog.py` are the places that already know everything a memlib entry needs.

If you would like to take it, a PR against `dev` would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch, and the regression suite is run with `cd compiler/tests && make -k -j<N>`.