VLSIDA/OpenRAM #166: Unit Tests - make -j 3

haotommy, 2023-01-05:
FAILING TESTS
freepdk45/20_psram_1bank_4mux_1rw_1r_test.bad
freepdk45/20_sram_1bank_2mux_1rw_1r_test.bad
freepdk45/20_sram_1bank_2mux_1w_1r_test.bad
freepdk45/20_sram_1bank_2mux_global_test.bad
freepdk45/20_sram_1bank_2mux_wmask_spare_cols_test.bad
freepdk45/20_sram_1bank_2mux_wmask_test.bad
freepdk45/20_sram_1bank_nomux_1rw_1r_spare_cols_test.bad
freepdk45/21_model_delay_test.bad
freepdk45/21_ngspice_delay_extra_rows_test.bad
freepdk45/21_ngspice_delay_test.bad
freepdk45/21_ngspice_setuphold_test.bad
freepdk45/21_xyce_delay_test.bad
freepdk45/21_xyce_setuphold_test.bad
freepdk45/23_lib_sram_test.bad
scn4m_subm/19_single_bank_global_bitline_test.bad
scn4m_subm/21_model_delay_test.bad
scn4m_subm/21_xyce_delay_test.bad
scn4m_subm/21_xyce_setuphold_test.bad
scn4m_subm/23_lib_sram_test.bad

why it is wrong?

The following URL is my openram installation process and results:
https://github.com/VLSIDA/OpenRAM/files/10350168/output.txt

I would like to ask if the installation was successful or there is an error?

haotommy, 2023-01-06:
commit 5ad1db95a80e04c541bf16d59678375a60734807 (HEAD -> stable, origin/stable, origin/HEAD)

haotommy, 2023-01-11:
> Can you specify which GitHub commit you are using? (First line of "git
> log")

Is this?
commit 5ad1db9
<https://github.com/VLSIDA/OpenRAM/commit/5ad1db95a80e04c541bf16d59678375a60734807>
(HEAD
-> stable, origin/stable, origin/HEAD)

mguthaus, 2026-10-16:
Closing as stale. No activity for about 3.8 years and no reproduction against current `dev`, so there is nothing actionable here as it stands.

The test harness has changed since. `compiler/tests/Makefile` is the entry point, parallelism is `make -k -jN`, and known-failing combinations are tracked explicitly in `BROKEN_STAMPS` rather than being silent.

To be clear this is closed for inactivity, not rejected. If it still affects you please reopen or file fresh, with a reproduction against current `dev` including your configuration file, technology, OS and `python3 --version`, and the temp-directory contents OpenRAM leaves on failure ([the FAQ lists which files](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself a PR against `dev` would be welcome and we will review it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md).

closed 2026-10-16 by mguthaus