VLSIDA/OpenRAM #169: Any document about analytical model used in OpenRAM for timing/area?

JiacongSun, 2023-01-13:
Hi,
I'm interested in what analytical model is used in OpenRAM for estimating latency and area of each memory component. I have checked the paper but it is not there. Is there any document, published paper or video that I can refer to? Appreciate.

Regards,
Jiacong

mguthaus, 2023-01-13:
Hi,

This would be the best place:
https://escholarship.org/content/qt7vx9n089/qt7vx9n089_noSplash_cfc4ba479d8eb1b6ec25d7c92357bc18.pdf?t=ra9wzr

I will create a list of all the papers in the docs as references before I close this issue.

mguthaus, 2023-01-13:
Updated the list in 5816c669f3966ba2ec046a87e1ada6a7b7ab6711

JiacongSun, 2023-01-13:
> Hi,
> 
> This would be the best place: https://escholarship.org/content/qt7vx9n089/qt7vx9n089_noSplash_cfc4ba479d8eb1b6ec25d7c92357bc18.pdf?t=ra9wzr
> 
> I will create a list of all the papers in the docs as references before I close this issue.

Awesome! Exactly what I am looking for. Thanks for your response!
Yes, I think that is great, thanks for your effort!

closed 2023-01-13 by JiacongSun