VLSIDA/OpenRAM #171: Dockerfile build error

rushikeshusc, 2023-01-16:
```
~/OpenRAM/docker$ make build
docker build -t vlsida/openram-ubuntu:2023-01-16 -f Dockerfile . | tee -i openram-ubuntu.log
[+] Building 2.6s (10/78)
 => [internal] load build definition from Dockerfile                                                     0.0s
 => => transferring dockerfile: 38B                                                                      0.0s
 => [internal] load .dockerignore                                                                        0.0s
 => => transferring context: 2B                                                                          0.0s
 => [internal] load metadata for docker.io/library/ubuntu:20.04                                          1.6s
 => [internal] load build context                                                                        0.1s
 => => transferring context: 63B                                                                         0.0s
 => [ 1/74] FROM docker.io/library/ubuntu:20.04@sha256:0e0402cd13f68137edb0266e1d2c682f217814420f2d43d3  0.0s
 => CACHED [ 2/74] RUN ln -s /usr/share/zoneinfo/America/Los_Angeles /etc/localtime                      0.0s
 => CACHED [ 3/74] RUN echo "America/Los_Angeles" > /etc/timezone                                        0.0s
 => CACHED [ 4/74] RUN apt-get update                                                                    0.0s
 => CACHED [ 5/74] RUN apt-get --no-install-recommends -y upgrade                                        0.0s
 => ERROR [ 6/74] RUN apt-get install --no-install-recommends -y build-essential git ssh vim gosu autoc  0.8s
------
 > [ 6/74] RUN apt-get install --no-install-recommends -y build-essential git ssh vim gosu autoconf automake libtool bison flex:
#9 0.559 Reading package lists...
#9 0.742 Building dependency tree...
#9 0.781 Reading state information...
#9 0.790 Package autoconf is not available, but is referred to by another package.
#9 0.790 This may mean that the package is missing, has been obsoleted, or
#9 0.790 is only available from another source
#9 0.790
#9 0.790 Package automake is not available, but is referred to by another package.
#9 0.790 This may mean that the package is missing, has been obsoleted, or
#9 0.790 is only available from another source
#9 0.790
#9 0.790 Package libtool is not available, but is referred to by another package.
#9 0.790 This may mean that the package is missing, has been obsoleted, or
#9 0.790 is only available from another source
#9 0.790
#9 0.791 E: Package 'autoconf' has no installation candidate
#9 0.791 E: Package 'automake' has no installation candidate
#9 0.791 E: Package 'libtool' has no installation candidate
#9 0.791 E: Unable to locate package bison
#9 0.791 E: Unable to locate package flex
------
executor failed running [/bin/sh -c apt-get install --no-install-recommends -y build-essential git ssh vim gosu autoconf automake libtool bison flex]: exit code: 100
docker tag  vlsida/openram-ubuntu:2023-01-16 vlsida/openram-ubuntu:latest
Error response from daemon: No such image: vlsida/openram-ubuntu:2023-01-16
make: *** [Makefile:11: build] Error 1
```

mguthaus, 2023-01-16:
This looks like a networking or system problem, not OpenRAM. Sorry, I don't know how to help more.

erendn, 2024-01-19:
We're now using Conda instead of Docker.

closed 2024-01-19 by erendn