VLSIDA/OpenRAM #174: way to fast estimate area?

JiacongSun, 2023-01-27:
Hello,

Anyone know how to fast estimate area for specific memory size? I found the process of generating a memory size bigger than 1kB takes more than 1 hour. How about if I just to want to get an estimated area value and no requirement of having a gds file. Is there any option can be used?

Kind regards,
Jiacong

mguthaus, 2023-01-27:
If you disable the power routing, disable DRV/LVS, and disable characterization it will be very quick.

route_supplies=False
check_lvsdrc=False
analytical_delay=True

closed 2023-01-27 by mguthaus

JiacongSun, 2023-01-27:
> If you disable the power routing, disable DRV/LVS, and disable characterization it will be very quick.
> 
> route_supplies=False check_lvsdrc=False analytical_delay=True

Thanks a lot!