VLSIDA/OpenRAM #176: Trimmed spice netlist only comments out the name of instance and not following port connections/subckt names on newlines

StephenMoreOSU, 2023-02-07:
**Describe the bug**
When generating the trimmed spice netlist, I believe the intention is to comment out the bitcells (including port connections and subckt definition) which have been trimmed out, however, when running the attached configuration file, the hspice simulation fails and prints the following error message:

Number of nodes miss match between instance "xbit_r0_c1      " and subcircuit "cell_1rw        ".  Subcircuit definition has    5 node(s) whereas subckt instance was found with   89 node(s). Please specify same number of nodes

This message makes sense as through the lack of comments on the ports and subckt definitions the cell xbit_r0_c1 expects all of them to be nodes for its circuit. 

**Version**
origin/stable

**To Reproduce**

Using the below configuration file (named SRAM_8x16_1rw.py), I ran the following command:

```bash
python3 $OPENRAM_HOME/../sram_compiler -v -c SRAM_8x16_1rw.py
```
The hspice simulation called on line 409 of OpenRAM/compiler/characterizer/stimuli.py fails and causes the program to abort.

The command hspice command that is run  and fails is
```bash
hspice -mt 3 -i <tmp/dir/path>/delay_stim.sp -o <tmp/dir/path>/timing
```

************ Configuration file starting here ************:

word_size = 8
num_words = 16

num_rw_ports = 1
num_r_ports = 0
num_w_ports = 0

tech_name = "freepdk45"
nominal_corner_only = True

check_lvsdrc = True

analytical_delay = True

spice_name = "hspice"

drc_name = "calibre"
lvs_name = "calibre"
pex_name = "calibre"

output_name = "sram_{0}rw{1}r{2}w_{3}_{4}_{5}".format(num_rw_ports,
                                                      num_r_ports,
                                                      num_w_ports,
                                                      word_size,
                                                      num_words,
                                                      tech_name)
output_path = "temp/{}".format(output_name)

************ Configuration Ending Here ************

**Expected behavior**

The hspice simulation called on line 409 of OpenRAM/compiler/characterizer/stimuli.py should return 0 and allow for characterization.

The command hspice command that should return 0 is
```bash
hspice -mt 3 -i <tmp/dir/path>/delay_stim.sp -o <tmp/dir/path>/timing
```

**Logs**
[image]
The above image is a screenshot of the trimmed.sp file which is written for the given configuration file and demonstrates the bug.

StephenMoreOSU, 2023-02-07:
Revision:

Version:
 v1.2.1

The fix for this issue is the change of the following lines in OpenRAM/compiler/base/heirarchy_spice.py :

[image]

mguthaus, 2023-02-07:
Ah, it looks like the line wrapping messed this up. We will get that in there ASAP.

bugraonal, 2023-07-20:
This should be fixed in the latest version.

closed 2023-07-20 by bugraonal