VLSIDA/OpenRAM #177: DRC and LVS errors in sky130_sram_4kbyte_1rw1r_32x1024_8

urielcho, 2023-02-13:
When I run sky130_sram_4kbyte_1rw1r_32x1024_8 it shows me 52 DRC errors and 3 LVS errors, but it was using the default settings just after installing OpenRAM

Git commit id: 31d2d7145b7714e3d1624aba50e9c2cce715adf7

in the macros folder use the following command

make sky130_sram_4kbyte_1rw1r_32x1024_8

Configuration file used

"""
Dual port (1 read/write + 1 read only), 4 kbytes SRAM (with byte write).

FIXME: What is this useful for?
FIXME: Why would you want byte write on this?
"""

word_size = 32 # Bits
num_words = 1024
human_byte_size = "{:.0f}kbytes".format((word_size * num_words)/1024/8)

# Allow byte writes
write_size = 8 # Bits

# Dual port
num_rw_ports = 1
num_r_ports = 1
num_w_ports = 0
ports_human = '1rw1r'

import os
exec(open(os.path.join(os.path.dirname(__file__), 'sky130_sram_common.py')).read())

I expected that the RAM will be generated without any error

mguthaus, 2023-02-13:
Thanks. We are aware of the DRC issue and don't yet have a solution. We did some hand-fixes on the macros that we taped out.

I'm a little surprised by the LVS error, though.  We will check it out.

urielcho, 2023-02-13:
Is it possible to manufacture with the Efabless MPW programs even with the DRC errors?

Instead of using 4kbyte memories, maybe I could use 1kbyte ones that don't have LVS errors but have 22 DRC errors, do you think that would be possible?

urielcho, 2023-02-13:
thank you very much that will be very useful.
I will be waiting for the memories of 4 and 8 kbytes of 8 word size if possible

closed 2023-02-13 by urielcho

manwu1994, 2024-03-22:
Hello, Dr. Maguthaus, does OpenRAM generate an SRAM with FreePDK following the below configuration:
```
word_size = 128
num_words = 256
write_size = 128

num_rw_ports = 0
num_r_ports = 1
num_w_ports = 1

tech_name = "freepdk45"
nominal_corner_only = True
```
Then, `ERROR: file signal_escape_router.py: line 70: Couldn't route from (addr1[4] layer=m3 ll=v[196.685,-1.2875] ur=v[196.82,-1.1525]) to (addr1[4] layer=m4 ll=v[196.6825,-26.16] ur=v[196.8225,-26.02]).`

Could you give me suggestions?

Thank you in advance.

mguthaus, 2024-03-22:
This is a very big data word for a small memory. Does it say if it's using a column mux or not?

manwu1994, 2024-03-23:
> This is a very big data word for a small memory. Does it say if it's using a column mux or not?

Thank you so much for your reply. We did not use column mux.

mguthaus, 2024-03-23:
I understand that you may not have used one, but openram does an internal calculation to figure out if it should use one. It is technology specific depending on the bitcell height. 

Can you include more of the log?
Can you open the resulting layout and see what is going on at those coordinates?

manwu1994, 2024-03-23:
Thank you for your response. I have attached the log regarding the parameters word_size = 128, num_words = 128, and write_size = 128, which experienced the same issue.

When attempting to generate the SRAM with a word size of 128 bits, it failed to generate the layout.

Furthermore, when generating the SRAM with a word size of 64 bits and 128 words, I successfully obtained the layout and parameters of the SRAM.
The below is the log of  `ERROR: file signal_escape_router.py: line 70: Couldn't route from (addr1[4] layer=m3 ll=v[196.685,-1.2875] ur=v[196.82,-1.1525]) to (addr1[4] layer=m4 ll=v[196.6825,-26.16] ur=v[196.8225,-26.02]).`
```
[2024-03-23 18:06:48.688639] |==============================================================================|
[2024-03-23 18:06:48.701868] |=========                      OpenRAM v1.2.42                       =========|
[2024-03-23 18:06:48.702460] |=========                                                            =========|
[2024-03-23 18:06:48.702886] |=========               VLSI Design and Automation Lab               =========|
[2024-03-23 18:06:48.703182] |=========        Computer Science and Engineering Department         =========|
[2024-03-23 18:06:48.703468] |=========            University of California Santa Cruz             =========|
[2024-03-23 18:06:48.703750] |=========                                                            =========|
[2024-03-23 18:06:48.704041] |=========          Usage help: openram-user-group@ucsc.edu           =========|
[2024-03-23 18:06:48.704344] |=========        Development help: openram-dev-group@ucsc.edu        =========|
[2024-03-23 18:06:48.704627] |=========                See LICENSE for license info                =========|
[2024-03-23 18:06:48.704921] |==============================================================================|
[2024-03-23 18:06:48.705222] ** Start: 03/23/2024 18:06:48
[2024-03-23 18:06:48.705513] Technology: freepdk45
[2024-03-23 18:06:48.705806] Total size: 16384 bits
[2024-03-23 18:06:48.706109] Word size: 128
Words: 128
Banks: 1
[2024-03-23 18:06:48.706411] RW ports: 0
R-only ports: 1
W-only ports: 1
[2024-03-23 18:06:48.706716] Design supply routing skipped. Supplies will have multiple must-connect pins. (route_supplies=True to enable supply routing).
[2024-03-23 18:06:48.707010] DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
[2024-03-23 18:06:48.707301] DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
[2024-03-23 18:06:48.707614] Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
[2024-03-23 18:06:48.707938] Only generating nominal corner timing.
[2024-03-23 18:06:48.708235] Words per row: None
[2024-03-23 18:06:48.708534] Output files are: 
[2024-03-23 18:06:48.708826] ../PIM/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_128_128_freepdk45/sram_0rw1r1w_128_128_freepdk45.lvs
[2024-03-23 18:06:48.709119] ../PIM/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_128_128_freepdk45/sram_0rw1r1w_128_128_freepdk45.sp
[2024-03-23 18:06:48.709420] ../PIM/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_128_128_freepdk45/sram_0rw1r1w_128_128_freepdk45.v
[2024-03-23 18:06:48.709711] ../PIM/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_128_128_freepdk45/sram_0rw1r1w_128_128_freepdk45.lib
[2024-03-23 18:06:48.710013] ../PIM/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_128_128_freepdk45/sram_0rw1r1w_128_128_freepdk45.py
[2024-03-23 18:06:48.710319] ../PIM/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_128_128_freepdk45/sram_0rw1r1w_128_128_freepdk45.html
[2024-03-23 18:06:48.710650] ../PIM/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_128_128_freepdk45/sram_0rw1r1w_128_128_freepdk45.log
[2024-03-23 18:06:48.710936] ..5/PIM/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_128_128_freepdk45/sram_0rw1r1w_128_128_freepdk45.lef
[2024-03-23 18:06:48.711230] ../PIM/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_128_128_freepdk45/sram_0rw1r1w_128_128_freepdk45.gds
[2024-03-23 18:09:01.711166] ** Submodules: 133.0 seconds
[2024-03-23 18:09:01.985917] ** Placement: 0.3 seconds
[2024-03-23 18:14:23.838512] ERROR: file signal_escape_router.py: line 70: Couldn't route from (addr1[4] layer=m3 ll=v[196.685,-1.2875] ur=v[196.82,-1.1525]) to (addr1[4] layer=m4 ll=v[196.6825,-26.16] ur=v[196.8225,-26.02]).
```

Thank you so much again.