VLSIDA/OpenRAM #179: Compilation of 4KB SRAM breaks with KeyError: 'nand4_leakage'

nklimov-dev, 2023-02-14:
_**Describe the bug**
A clear and concise description of what the bug is._

The compilation of a 4x1024 SRAM breaks with an error:

**KeyError: 'nand4_leakage'**

yielded from:

**File "/openram/compiler/modules/pnand4.py", line 328, in analytical_power
    power_leak = spice["nand4_leakage"]**

(Please refer to log output below).

_**Version**
Which commit are you using?_

Commit 31d2d71

_**To Reproduce**
What did you do to demonstrate the bug?
Please include your configuration file used._

After cloning the OpenRAM repository, I entered the repository folder and ran "sudo make library", then "sudo make pdk", and then "sudo make install". Then, I went to the docker/ directory and ran "sudo make build". Then, I ran "sudo make mount". Inside the Docker container, I cloned the OpenRAM repository (to obtain the compiler), and ran the command "python3 $OPENRAM_ROOT/sram_compiler.py sram_4_1024_sky130.py". The contents of the configuration file "sram_4_1024_sky130.py" is provided below:

**# Data word size
word_size = 4
# Number of words in the memory
num_words = 1024

# Technology to use in $OPENRAM_TECH
tech_name = "sky130"

# You can use the technology nominal corner only
nominal_corner_only = True
# Or you can specify particular corners
# Process corners to characterize
process_corners = ["TT"]
# Voltage corners to characterize
supply_voltages = [3.3]
# Temperature corners to characterize
temperatures = [ 0, 25, 100]

num_rw_ports = 2
num_r_ports = 0
num_w_ports = 0

# Output directory for the results
output_path = "temp"
# Output file base name
output_name = "sram_{0}_{1}_{2}".format(word_size,num_words,tech_name)

# Disable analytical models for full characterization (WARNING: slow!)
# analytical_delay = False**

_**Expected behavior**
A clear and concise description of what you expected to happen._

I was expecting for the compilation to finish and yield all of the output files. I did obtain the following files:
functional_stim.sp
sram_4_1024_sky130.lvs
sram_4_1024_sky130.sp
sram_4_1024_sky130.lib
sram_4_1024_sky130.log
sram_4_1024_sky130.lef
sram_4_1024_sky130.gds

However, I did not obtain the following output files:
sram_4_1024_sky130.v
sram_4_1024_sky130.py
[... 606 lines cut ...]
  File "/openram/compiler/characterizer/lib.py", line 163, in characterize_corners
    self.characterize()
  File "/openram/compiler/characterizer/lib.py", line 175, in characterize
    self.compute_delay()
  File "/openram/compiler/characterizer/lib.py", line 646, in compute_delay
    char_results = m.get_lib_values(self.load_slews)
  File "/openram/compiler/characterizer/elmore.py", line 63, in get_lib_values
    power = self.analytical_power(load_slews)
  File "/openram/compiler/characterizer/elmore.py", line 103, in analytical_power
    power = self.sram.analytical_power(self.corner, load)
  File "/openram/compiler/base/design.py", line 127, in analytical_power
    total_module_power += inst.mod.analytical_power(corner, load)
  File "/openram/compiler/base/design.py", line 127, in analytical_power
    total_module_power += inst.mod.analytical_power(corner, load)
  File "/openram/compiler/base/design.py", line 127, in analytical_power
    total_module_power += inst.mod.analytical_power(corner, load)
  [Previous line repeated 2 more times]
  File "/openram/compiler/modules/pnand4.py", line 328, in analytical_power
    power_leak = spice["nand4_leakage"]
KeyError: 'nand4_leakage'**

The log file has the same contents as the console output.

**Additional context**
Add any other context about the problem here.

kirbyydoge, 2023-02-16:
Add `spice["nand4_leakage"] = 1` to `technology/sky130/tech.py`.

mguthaus, 2026-10-13:
Closing as stale. This has been inactive for 3.7 years and there is no current reproduction against `dev`, so it is not actionable as it stands.

The decoder and module naming have changed considerably since; a `KeyError` on `nand4_l` may no longer reproduce.

To be clear, this is being closed for inactivity rather than rejected. If it still affects you, please reopen or file a fresh issue with a reproduction against current `dev`, including your configuration file, technology, OS and `python3 --version`, and the contents of the temp directory OpenRAM leaves behind on failure ([the FAQ explains which files those are](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself, a PR would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch.

closed 2026-10-13 by mguthaus