VLSIDA/OpenRAM #180: University Project - Porting to a new technology

rushikeshusc, 2023-02-16:
Hi, I'm a student at USC, and I'm working with a professor on a project where we were hoping to utilize OpenRAM. I see that you have guidelines for what might be required to port OpenRAM to a new technology, but I haven't found any examples of people who have successfully done it.

I was wondering if it is possible to port OpenRAM to a new PDK, and if yes, how would you define the complexity of such a project? Can you point me to any teams who have succesfully ported to a new technology not currently supported directly by the OpenRAM team? Thank you.

mguthaus, 2023-02-16:
Hi Rushikesh,

This is a big project. I would say it is a "large" MS project or a good 3-4 person undergrad project. This partly depends on the technology and how different it is from our existing ones. There may be some features that need to be added to OpenRAM for FinFET, for examle, or other new things. 

Hope that helps.

Matt

mguthaus, 2023-02-16:
The documentation is here:

https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/index.md

closed 2023-02-16 by mguthaus