VLSIDA/OpenRAM #182: Could not find the "fet_libraries" or "fet_models" during the compilation.

apaul33, 2023-03-01:
**Describe the bug**
A clear and concise description of what the bug is.

While running the compiler for an individual PDK, the compiler could not get the fet libraries or models that has defined on the each individual PDK's technology directory.

**KeyError:** AssertionError
 ERROR: file stimuli.py: line 56: Must define either fet_libraries or fet_models.

**yielded from** : While srart to run .sp files /home/azmir/OpenRAM/compiler/characterizer/stimuli.py
Predicted to give error while start to generate other output files.
Please refer to the log output given below, for better understanding.

**Version**: 1.2.1

**To Reproduce**
What did you do to demonstrate the bug?
Please include your configuration file used.

I checked the tech.py file on the PDK's tech directory to check whether fet_libraries or fet_models is defined or not. It is defined perfectly. Thus, I couldn't understand from where the error occurs.

**Myconfig File:**

#Data word size
word_size = 2
#Number of words in the memory
num_words = 16
num_banks=1

#Technology to use in $OPENRAM_TECH
tech_name ="sky130"
#You can use the technology nominal corner only
#nominal_corner_only = True
#Or you can specify particular corners
#Process corners to characterize
process_corners = ["SS", "TT", "FF"]
#Voltage corners to characterize
supply_voltages = [ 3.0, 3.3, 3.5 ]
#Temperature corners to characterize
temperatures = [ 0, 25, 100]

#Output directory for the results
output_path = "sky_130_check"
#Output file base name
output_name = "sram_{0}_{1}_{2}".format(word_size,num_words,tech_name)

#Disable analytical models for full characterization (WARNING: slow!)
#analytical_delay = False

**Expected behavior**
I am expecting to generate the below files after competing the compilation

Output files:
sram_2_16_sky130.lvs
sram_2_16_sky130.sp
sram_2_16_sky130.v
sram_2_16_sky130.lib
sram_2_16_sky130.py
sram_2_16_sky130.html
sram_2_16_sky130.log
sram_2_16_sky130.lef
sram_2_16_sky130.gds

The compiler can only generate the log file. |
As soon as the .sp file start generating the error occurs.

**Logs:**

azmir@compiler:~/OpenRAM$ python3 sram_compiler.py myconfig.py 
|==============================================================================|
|=========                       OpenRAM v1.2.1                       =========|
[... 88 lines cut ...]
ERROR: file bitcell_base.py: line 273: Must override build_graph function in bitcell base class.
ERROR: file bitcell_base.py: line 273: Must override build_graph function in bitcell base class.
ERROR: file bitcell_base.py: line 273: Must override build_graph function in bitcell base class.
ERROR: file bitcell_base.py: line 273: Must override build_graph function in bitcell base class.
ERROR: file bitcell_base.py: line 273: Must override build_graph function in bitcell base class.
ERROR: file bitcell_base.py: line 273: Must override build_graph function in bitcell base class.
ERROR: file bitcell_base.py: line 273: Must override build_graph function in bitcell base class.
ERROR: file bitcell_base.py: line 273: Must override build_graph function in bitcell base class.
ERROR: file bitcell_base.py: line 273: Must override build_graph function in bitcell base class.
ERROR: file bitcell_base.py: line 273: Must override build_graph function in bitcell base class.
ERROR: file stimuli.py: line 52: Must define either fet_libraries or fet_models.
Traceback (most recent call last):
  File "/home/azmir/OpenRAM/sram_compiler.py", line 83, in <module>
    s.save()
  File "/home/azmir/OpenRAM/compiler/sram.py", line 104, in save
    functional(self.s,
  File "/home/azmir/OpenRAM/compiler/characterizer/functional.py", line 102, in __init__
    self.write_functional_stimulus()
  File "/home/azmir/OpenRAM/compiler/characterizer/functional.py", line 386, in write_functional_stimulus
    self.stim = stimuli(self.sf, self.corner)
  File "/home/azmir/OpenRAM/compiler/characterizer/stimuli.py", line 52, in __init__
    debug.error("Must define either fet_libraries or fet_models.", -1)
  File "/home/azmir/OpenRAM/compiler/debug.py", line 47, in error
    assert return_value == 0
AssertionError

mguthaus, 2023-03-01:
I had to add this to your config to get a valid width/height of the array:
num_spare_rows = 1
num_spare_cols = 1

It seems that we don't map the corners in the technology file except for TT. You can work around this with:
nominal_corner_only = True
for now and I will work on getting these added as a fix.

mguthaus, 2026-10-13:
Closing as stale. This has been inactive for 3.6 years and there is no current reproduction against `dev`, so it is not actionable as it stands.

This is usually a PDK setup problem rather than a code bug — see [basic_setup.md](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/basic_setup.md) for the required environment variables and PDK install steps.

To be clear, this is being closed for inactivity rather than rejected. If it still affects you, please reopen or file a fresh issue with a reproduction against current `dev`, including your configuration file, technology, OS and `python3 --version`, and the contents of the temp directory OpenRAM leaves behind on failure ([the FAQ explains which files those are](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself, a PR would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch.

closed 2026-10-13 by mguthaus