VLSIDA/OpenRAM #183: ERROR: file router.py: line 702: No pins overlapped the tracks.

zpzhu99, 2023-03-07:
**Describe the bug**
A clear and concise description of what the bug is.
I am generating a dual port (1r1w) SRAM with sky130 pdk. An error saying "ERROR: file router.py: line 702: No pins overlapped the tracks" happens after "Finding pins and blockages: 245.5 seconds". The same error won't occur for a single port SRAM. What could have caused the problem? Thanks!

**Version**
Which commit are you using?
V1.2.1
31d2d7145b7714e3d1624aba50e9c2cce715adf7

**To Reproduce**
What did you do to demonstrate the bug?
Please include your configuration file used.

I have two configuration files for the SRAM, similar to the given macros for sky130 SRAMs.

```
word_size = 32 # Bits
num_words = 256
human_byte_size = "{:.0f}kbytes".format((word_size * num_words)/1024/8)

write_size = 8 # Bits

num_rw_ports = 0
num_r_ports = 1
num_w_ports = 1
num_spare_rows = 0
num_spare_cols = 0
ports_human = '1r1w'
esp_name = "esp_aux_dual"

import os
exec(open(os.path.join(os.path.dirname(__file__), 'sky130_sram_common.py')).read())
```

```
tech_name = "sky130"
nominal_corner_only = True

# Local wordlines have issues with met3 power routing for now
#local_array_size = 16

route_supplies = "ring"
#route_supplies = "left"
check_lvsdrc = False
uniquify = True
#perimeter_pins = False
#netlist_only = True
#analytical_delay = False

num_threads = 4

output_name = "{tech_name}_{esp_name}_sram_{human_byte_size}_{ports_human}_{word_size}x{num_words}_{write_size}".format(**locals())
output_path = "macro/{output_name}".format(**locals())
```

**Expected behavior**

**Logs**
If applicable, add logs or output to help explain your problem.

```
|==============================================================================|
|=========                       OpenRAM v1.2.1                       =========|
|=========                                                            =========|
|=========               VLSI Design and Automation Lab               =========|
|=========        Computer Science and Engineering Department         =========|
|=========            University of California Santa Cruz             =========|
|=========                                                            =========|
|=========          Usage help: openram-user-group@ucsc.edu           =========|
|=========        Development help: openram-dev-group@ucsc.edu        =========|
|=========                See LICENSE for license info                =========|
|==============================================================================|
** Start: 03/07/2023 12:35:59
Technology: sky130
Total size: 8192 bits
Word size: 32
Words: 256
Banks: 1
Write size: 8
RW ports: 0
R-only ports: 1
W-only ports: 1
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
Only generating nominal corner timing.
Words per row: None
Output files are: 
/OpenRAM/esp_srams/macro/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.lvs
/OpenRAM/esp_srams/macro/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.sp
/OpenRAM/esp_srams/macro/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.v
/OpenRAM/esp_srams/macro/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.lib
/OpenRAM/esp_srams/macro/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.py
/OpenRAM/esp_srams/macro/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.html
/OpenRAM/esp_srams/macro/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.log
/OpenRAM/esp_srams/macro/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.lef
/OpenRAM/esp_srams/macro/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.gds
WARNING: file hierarchy_layout.py: line 642: Could not find pin gnd on col_cap_bitcell_2port
WARNING: file hierarchy_layout.py: line 642: Could not find pin gnd on col_cap_bitcell_2port
WARNING: file hierarchy_layout.py: line 642: Could not find pin gnd on col_cap_bitcell_2port
WARNING: file hierarchy_layout.py: line 642: Could not find pin gnd on col_cap_bitcell_2port
** Submodules: 11.2 seconds
** Placement: 0.1 seconds
**** Retrieving pins: 0.0 seconds
**** Analyzing pins: 0.0 seconds
**** Finding blockages: 11.7 seconds
**** Converting blockages: 1.1 seconds
**** Converting pins: 0.4 seconds
**** Separating adjacent pins: 0.1 seconds
*** Finding pins and blockages: 128.8 seconds
*** Maze routing pins: 313.0 seconds
**** Retrieving pins: 0.0 seconds
**** Analyzing pins: 0.4 seconds
**** Finding blockages: 24.7 seconds
**** Converting blockages: 1.1 seconds
**** Converting pins: 13.7 seconds
**** Separating adjacent pins: 89.0 seconds
*** Finding pins and blockages: 245.5 seconds
ERROR: file router.py: line 702: No pins overlapped the tracks.
Traceback (most recent call last):
  File "../sram_compiler.py", line 80, in <module>
    s = sram()
  File "/OpenRAM/compiler/sram.py", line 58, in __init__
    self.s.create_layout()
  File "/OpenRAM/compiler/modules/sram_1bank.py", line 216, in create_layout
    self.route_layout()
  File "/OpenRAM/compiler/modules/sram_1bank.py", line 1093, in route_layout
    self.route_supplies(init_bbox)
  File "/OpenRAM/compiler/modules/sram_1bank.py", line 265, in route_supplies
    rtr.route()
  File "/OpenRAM/compiler/router/supply_tree_router.py", line 86, in route
    self.route_pins(gnd_name)
  File "/OpenRAM/compiler/router/supply_tree_router.py", line 152, in route_pins
    self.route_signal(pin_name, src, dest)
  File "/OpenRAM/compiler/router/supply_tree_router.py", line 191, in route_signal
    if self.run_router(detour_scale=detour_scale):
  File "OpenRAM/compiler/router/router.py", line 1320, in run_router
    self.target_components)
  File "/OpenRAM/compiler/router/router.py", line 1199, in create_route_connector
    partial_pin_parts = self.divide_pin_to_tracks(pin, pg.grids)
  File "/OpenRAM/compiler/router/router.py", line 702, in divide_pin_to_tracks
    debug.check(len(overlap_pins) > 0, "No pins overlapped the tracks.")
  File "/OpenRAM/compiler/debug.py", line 33, in check
    assert 0
AssertionError
```

**Additional context**
Add any other context about the problem here.

andreili, 2023-03-31:
Have a same problem.
Config:
`num_banks=2
word_size = 32
num_words = 8192
write_size = 8

local_array_size = 32

num_rw_ports = 2
num_r_ports = 0
num_w_ports = 0

tech_name = "sky130"
#nominal_corner_only = True
process_corners = [ "TT" ]
supply_voltages = [ 1.8 ]
temperatures = [ 25 ]

#route_supplies = False
#check_lvsdrc = False
#perimeter_pins = False
#netlist_only = True
#analytical_delay = False

drc_name = "magic"
lvs_name = "netgen"
pex_name = "magic"

output_name = "{0}_sram_{1}rw{2}r{3}w_{4}_{5}".format(tech_name,
                                                      num_rw_ports,
                                                      num_r_ports,
                                                      num_w_ports,
                                                      word_size,
                                                      num_words)
output_path = "{}".format(output_name)
verbose_level = 1 `
Log:
`(venv) (base) andreil@andreil /media/ssd/ram_ex $ ${OPENRAM_HOME}/../sram_compiler.py sky130_sram
[openram.globals/read_config]: Output saved in /media/ssd/ram_ex/sky130_sram_2rw0r0w_32_8192/
[openram.globals/install_conda]: Creating conda setup...
[openram.globals/import_tech]: Tech directory found in /media/ssd/OpenRAM/technology:/media/ssd/OpenRAM/compiler/../technology
[openram.globals/import_tech]: Adding technology path: /media/ssd/OpenRAM/technology
[openram.globals/import_tech]: Adding technology path: /media/ssd/OpenRAM/compiler/../technology
[openram.globals/init_paths]: Creating temp directory: /tmp/openram_andreil_431950_temp/
[openram.globals/setup_bitcell]: Using bitcell: bitcell_2port
[openram.characterizer/<module>]: Initializing characterizer...
[openram.characterizer/<module>]: Analytical model enabled.
[openram.verify/<module>]: Initializing verify...
[openram.verify/<module>]: LVS/DRC/PEX disabled.
[openram.globals/setup_bitcell]: Using bitcell: bitcell_2port
|==============================================================================|
|=========                       OpenRAM v1.2.7                       =========|
|=========                                                            =========|
|=========               VLSI Design and Automation Lab               =========|
|=========        Computer Science and Engineering Department         =========|
|=========            University of California Santa Cruz             =========|
|=========                                                            =========|
|=========          Usage help: openram-user-group@ucsc.edu           =========|
|=========        Development help: openram-dev-group@ucsc.edu        =========|
|=========                See LICENSE for license info                =========|
|==============================================================================|
** Start: 03/30/2023 10:22:18
Technology: sky130
Total size: 524288 bits
Word size: 32
Words: 8192
Banks: 2
Write size: 8
RW ports: 2
[... 343 lines cut ...]
  File "/media/ssd/OpenRAM/compiler/../sram_compiler.py", line 71, in <module>
    s = sram()
  File "/media/ssd/OpenRAM/compiler/sram.py", line 58, in __init__
    self.s.create_layout()
  File "/media/ssd/OpenRAM/compiler/modules/sram_1bank.py", line 216, in create_layout
    self.route_layout()
  File "/media/ssd/OpenRAM/compiler/modules/sram_1bank.py", line 1093, in route_layout
    self.route_supplies(init_bbox)
  File "/media/ssd/OpenRAM/compiler/modules/sram_1bank.py", line 265, in route_supplies
    rtr.route()
  File "/media/ssd/OpenRAM/compiler/router/supply_tree_router.py", line 86, in route
    self.route_pins(gnd_name)
  File "/media/ssd/OpenRAM/compiler/router/supply_tree_router.py", line 152, in route_pins
    self.route_signal(pin_name, src, dest)
  File "/media/ssd/OpenRAM/compiler/router/supply_tree_router.py", line 191, in route_signal
    if self.run_router(detour_scale=detour_scale):
  File "/media/ssd/OpenRAM/compiler/router/router.py", line 1318, in run_router
    self.create_route_connector(path,
  File "/media/ssd/OpenRAM/compiler/router/router.py", line 1199, in create_route_connector
    partial_pin_parts = self.divide_pin_to_tracks(pin, pg.grids)
  File "/media/ssd/OpenRAM/compiler/router/router.py", line 702, in divide_pin_to_tracks
    debug.check(len(overlap_pins) > 0, "No pins overlapped the tracks.")
  File "/media/ssd/OpenRAM/compiler/debug.py", line 33, in check
    assert 0
AssertionError`

jamesbowman, 2023-04-05:
Confirmed (using the first config above) that there is still the same failure at v1.2.6:

```
|==============================================================================|
|=========                       OpenRAM v1.2.6                       =========|
|=========                                                            =========|
|=========               VLSI Design and Automation Lab               =========|
|=========        Computer Science and Engineering Department         =========|
|=========            University of California Santa Cruz             =========|
|=========                                                            =========|
|=========          Usage help: openram-user-group@ucsc.edu           =========|
|=========        Development help: openram-dev-group@ucsc.edu        =========|
|=========                See LICENSE for license info                =========|
|==============================================================================|
** Start: 04/04/2023 14:51:59
Technology: sky130
Total size: 8192 bits
Word size: 32
Words: 256
Banks: 1
Write size: 8
RW ports: 0
R-only ports: 1
W-only ports: 1
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
Only generating nominal corner timing.
Words per row: None
Output files are:
/home/jamesb/OpenRAM/temp/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.lvs
/home/jamesb/OpenRAM/temp/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.sp
/home/jamesb/OpenRAM/temp/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.v
/home/jamesb/OpenRAM/temp/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.lib
/home/jamesb/OpenRAM/temp/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.py
/home/jamesb/OpenRAM/temp/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.html
/home/jamesb/OpenRAM/temp/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.log
/home/jamesb/OpenRAM/temp/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.lef
/home/jamesb/OpenRAM/temp/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8/sky130_esp_aux_dual_sram_1kbytes_1r1w_32x256_8.gds
** Submodules: 4.3 seconds
** Placement: 0.0 seconds
**** Retrieving pins: 0.0 seconds
**** Analyzing pins: 0.0 seconds
**** Finding blockages: 2.1 seconds
**** Converting blockages: 0.2 seconds
**** Converting pins: 0.2 seconds
**** Separating adjacent pins: 0.0 seconds
*** Finding pins and blockages: 26.4 seconds
*** Maze routing pins: 116.4 seconds
**** Retrieving pins: 0.0 seconds
**** Analyzing pins: 0.2 seconds
**** Finding blockages: 10.5 seconds
**** Converting blockages: 0.4 seconds
**** Converting pins: 7.9 seconds
**** Separating adjacent pins: 43.3 seconds
*** Finding pins and blockages: 118.1 seconds
ERROR: file router.py: line 702: No pins overlapped the tracks.
```

jamesbowman, 2023-04-05:
Second case takes a really long time, but ends with:

```
[openram.router.supply_tree_router/route_pins]: Routing gnd with 123 pins.
[openram.router.supply_tree_router/route_pins]: 0 supply segments routed, 122 remaining.
[openram.router.supply_tree_router/route_pins]: 25 supply segments routed, 97 remaining.
ERROR: file router.py: line 702: No pins overlapped the tracks.
```

mguthaus, 2023-04-06:
Hi James,

We met at latch up, right?

Classes are just getting started and this is probably not a simple fix so we will try to get to it soon...

erendn, 2023-10-12:
We have implemented a new router from scratch that might have fixed this issue. Could you all try running these again?

andreili, 2023-10-12:
Hi, @erendn 
Ok, I will try run it.

andreili, 2023-10-16:
First configuration - all ok, w/o error.
P.S.: I apologize for the delay, the electricity was turned off at night, nothing was saved. I had to restart when I was at home.

```
|==============================================================================|
|=========                      OpenRAM v1.2.40                       =========|
|=========                                                            =========|
|=========               VLSI Design and Automation Lab               =========|
|=========        Computer Science and Engineering Department         =========|
|=========            University of California Santa Cruz             =========|
|=========                                                            =========|
|=========          Usage help: openram-user-group@ucsc.edu           =========|
|=========        Development help: openram-dev-group@ucsc.edu        =========|
|=========                See LICENSE for license info                =========|
|==============================================================================|
** Start: 10/16/2023 08:36:45
Technology: sky130
Total size: 8192 bits
Word size: 32
Words: 256
Banks: 1
Write size: 8
RW ports: 0
R-only ports: 1
W-only ports: 1
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
Only generating nominal corner timing.
Words per row: None
Output files are:
/home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.lvs
/home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.sp
/home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.v
/home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.lib
/home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.py
/home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.html
/home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.log
/home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.lef
/home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.gds
** Submodules: 19.5 seconds
** Placement: 0.0 seconds
** Routing: 1010.7 seconds
** Verification: 11244.9 seconds
** SRAM creation: 12275.2 seconds
SP: Writing to /home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.sp
** Spice writing: 0.1 seconds
DELAY: Writing stimulus...
** DELAY: 0.6 seconds
GDS: Writing to /home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.gds
** GDS: 0.5 seconds
LEF: Writing to /home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.lef
** LEF: 0.0 seconds
LVS: Writing to /home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.lvs.sp
** LVS writing: 0.1 seconds
LIB: Characterizing...
WARNING: file elmore.py: line 45: In analytical mode, all ports have the timing of the first read port.
** Characterization: 1.0 seconds
Config: Writing to /home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.py
** Config: 0.0 seconds
Datasheet: Writing to /home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.html
** Datasheet: 0.0 seconds
Verilog: Writing to /home/andreil/ASIC/ram_ex/macro/sky130_sram_1kbytes_1r1w_32x256_8/sky130_sram_1kbytes_1r1w_32x256_8.v
** Verilog: 0.0 seconds
** End: 12277.6 seconds
```

erendn, 2023-10-17:
This issue seems to be fixed.

closed 2023-10-17 by erendn