VLSIDA/OpenRAM #189: Got TypeError when running example

jiegec, 2023-04-08:
**Describe the bug**

Got error when running example `python3 ../../sram_compiler.py example_config_1w_1r_scn4m_subm.py`:

```
Traceback (most recent call last):
  File "/home/jiegec/OpenRAM/macros/sram_configs/../../sram_compiler.py", line 74, in <module>
    s.save()
  File "/home/jiegec/OpenRAM/compiler/sram.py", line 104, in save
    functional(self.s,
  File "/home/jiegec/OpenRAM/compiler/characterizer/functional.py", line 99, in __init__
    self.create_random_memory_sequence()
  File "/home/jiegec/OpenRAM/compiler/characterizer/functional.py", line 152, in create_random_memory_sequence
    addr = self.gen_addr()
           ^^^^^^^^^^^^^^^
  File "/home/jiegec/OpenRAM/compiler/characterizer/functional.py", line 366, in gen_addr
    random_value = random.sample(self.valid_addresses, 1)[0]
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/random.py", line 439, in sample
    raise TypeError("Population must be a sequence.  "
TypeError: Population must be a sequence.  For dicts or sets, use sorted(d).
```

**Version**
commit 85db2043b27f842f8b927fe1a49deb42b0b05b40

**To Reproduce**
What did you do to demonstrate the bug?
Please include your configuration file used.

```shell
$ cd macros/sram_configs
$ python3 ../../sram_compiler.py example_config_1w_1r_scn4m_subm.py
WARNING: file globals.py: line 180: Failed to find coverage installation. This can be installed with pip3 install coverage
|==============================================================================|
|=========                       OpenRAM v1.2.8                       =========|
|=========                                                            =========|
|=========               VLSI Design and Automation Lab               =========|
|=========        Computer Science and Engineering Department         =========|
|=========            University of California Santa Cruz             =========|
|=========                                                            =========|
|=========          Usage help: openram-user-group@ucsc.edu           =========|
|=========        Development help: openram-dev-group@ucsc.edu        =========|
|=========                See LICENSE for license info                =========|
|==============================================================================|
** Start: 04/08/2023 09:38:53
Technology: scn4m_subm
Total size: 32 bits
Word size: 2
Words: 16
Banks: 1
RW ports: 0
R-only ports: 1
W-only ports: 1
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
Words per row: None
Output files are:
/home/jiegec/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_2_16_scn4m_subm/sram_0rw1r1w_2_16_scn4m_subm.lvs
/home/jiegec/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_2_16_scn4m_subm/sram_0rw1r1w_2_16_scn4m_subm.sp
/home/jiegec/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_2_16_scn4m_subm/sram_0rw1r1w_2_16_scn4m_subm.v
/home/jiegec/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_2_16_scn4m_subm/sram_0rw1r1w_2_16_scn4m_subm.lib
/home/jiegec/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_2_16_scn4m_subm/sram_0rw1r1w_2_16_scn4m_subm.py
/home/jiegec/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_2_16_scn4m_subm/sram_0rw1r1w_2_16_scn4m_subm.html
/home/jiegec/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_2_16_scn4m_subm/sram_0rw1r1w_2_16_scn4m_subm.log
/home/jiegec/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_2_16_scn4m_subm/sram_0rw1r1w_2_16_scn4m_subm.lef
/home/jiegec/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_2_16_scn4m_subm/sram_0rw1r1w_2_16_scn4m_subm.gds
** Submodules: 0.3 seconds
** Placement: 0.0 seconds
**** Retrieving pins: 0.0 seconds
**** Analyzing pins: 0.0 seconds
**** Finding blockages: 0.0 seconds
**** Converting blockages: 0.0 seconds
**** Converting pins: 0.0 seconds
**** Separating adjacent pins: 0.0 seconds
*** Finding pins and blockages: 0.6 seconds
*** Maze routing pins: 2.3 seconds
**** Retrieving pins: 0.0 seconds
**** Analyzing pins: 0.0 seconds
**** Finding blockages: 0.1 seconds
**** Converting blockages: 0.0 seconds
**** Converting pins: 0.7 seconds
**** Separating adjacent pins: 0.4 seconds
*** Finding pins and blockages: 1.7 seconds
*** Maze routing supplies: 13.5 seconds
** Routing: 24.4 seconds
** Verification: 4.0 seconds
** SRAM creation: 28.7 seconds
SP: Writing to /home/jiegec/OpenRAM/macros/sram_configs/macro/sram_0rw1r1w_2_16_scn4m_subm/sram_0rw1r1w_2_16_scn4m_subm.sp
Traceback (most recent call last):
  File "/home/jiegec/OpenRAM/macros/sram_configs/../../sram_compiler.py", line 74, in <module>
    s.save()
  File "/home/jiegec/OpenRAM/compiler/sram.py", line 104, in save
    functional(self.s,
  File "/home/jiegec/OpenRAM/compiler/characterizer/functional.py", line 99, in __init__
    self.create_random_memory_sequence()
  File "/home/jiegec/OpenRAM/compiler/characterizer/functional.py", line 152, in create_random_memory_sequence
    addr = self.gen_addr()
           ^^^^^^^^^^^^^^^
  File "/home/jiegec/OpenRAM/compiler/characterizer/functional.py", line 366, in gen_addr
    random_value = random.sample(self.valid_addresses, 1)[0]
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/random.py", line 439, in sample
    raise TypeError("Population must be a sequence.  "
TypeError: Population must be a sequence.  For dicts or sets, use sorted(d).
```

**Expected behavior**
A clear and concise description of what you expected to happen.

It should work.

**Logs**
If applicable, add logs or output to help explain your problem.

Pasted above.

**Additional context**
Add any other context about the problem here.

It can be fixed via a small change:

```diff
diff --git a/compiler/characterizer/functional.py b/compiler/characterizer/functional.py
index 78a44530..ddf0d588 100644
--- a/compiler/characterizer/functional.py
+++ b/compiler/characterizer/functional.py
@@ -363,7 +363,7 @@ class functional(simulation):
     def gen_addr(self):
         """ Generates a random address value to write to. """
         if self.valid_addresses:
-            random_value = random.sample(self.valid_addresses, 1)[0]
+            random_value = random.sample(list(self.valid_addresses), 1)[0]
         else:
             random_value = random.randint(0, self.max_address)
         addr_bits = binary_repr(random_value, self.bank_addr_size)
```

mguthaus, 2023-04-08:
Which version of python are you using? It might be too new.

mguthaus, 2023-04-08:
We will include this fixed though, thank you

jiegec, 2023-04-08:
> Which version of python are you using? It might be too new.

Python 3.11

samuelkcrow, 2023-05-05:
This should be fixed by this commit: 0e781dd2247b2440235f39ed9169cf0d42948ae7
Let me know if this does not resolve your issue.

closed 2023-05-05 by samuelkcrow