VLSIDA/OpenRAM #190: ERROR:Replica bit cannot be the dummy row.

akifcelik, 2023-04-08:
Hi, I tried to start  flow a simple sram with simple configuration and encountered error as following:

 -------------------------------------
ERROR: file sky130_replica_column.py: line 51: Replica bit cannot be the dummy row.
Traceback (most recent call last):
  File "../OpenRAM/sram_compiler.py", line 71, in <module>
    s = sram()
  File "/home/akif/OpenRAM/compiler/sram.py", line 56, in __init__
    self.s.create_netlist()
  File "/home/akif/OpenRAM/compiler/modules/sram_1bank.py", line 197, in create_netlist
    self.add_modules()
  File "/home/akif/OpenRAM/compiler/modules/sram_1bank.py", line 463, in add_modules
    self.bank = factory.create("bank", sram_config=self.sram_config, module_name="bank")
  File "/home/akif/OpenRAM/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
  File "/home/akif/OpenRAM/compiler/modules/bank.py", line 49, in __init__
    self.create_netlist()
  File "/home/akif/OpenRAM/compiler/modules/bank.py", line 59, in create_netlist
    self.add_modules()
  File "/home/akif/OpenRAM/compiler/modules/bank.py", line 377, in add_modules
    self.bitcell_array = factory.create(module_type="capped_replica_bitcell_array",
  File "/home/akif/OpenRAM/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
  File "/home/akif/OpenRAM/compiler/modules/capped_replica_bitcell_array.py", line 56, in __init__
    self.create_netlist()
  File "/home/akif/OpenRAM/compiler/modules/capped_replica_bitcell_array.py", line 62, in create_netlist
    self.add_modules()
  File "/home/akif/OpenRAM/compiler/modules/capped_replica_bitcell_array.py", line 90, in add_modules
    self.replica_bitcell_array = factory.create(module_type="replica_bitcell_array",
  File "/home/akif/OpenRAM/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
  File "/home/akif/OpenRAM/technology/sky130/custom/sky130_replica_bitcell_array.py", line 61, in __init__
    super().__init__(self.row_size, self.column_size, rbl, left_rbl, right_rbl, name)
  File "/home/akif/OpenRAM/compiler/modules/replica_bitcell_array.py", line 56, in __init__
    self.create_netlist()
  File "/home/akif/OpenRAM/compiler/modules/replica_bitcell_array.py", line 62, in create_netlist
    self.add_modules()
  File "/home/akif/OpenRAM/compiler/modules/replica_bitcell_array.py", line 116, in add_modules
    self.replica_columns[port] = factory.create(module_type="replica_column",
  File "/home/akif/OpenRAM/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
  File "/home/akif/OpenRAM/technology/sky130/custom/sky130_replica_column.py", line 51, in __init__
    debug.check(replica_bit != 0 and replica_bit != rows,
  File "/home/akif/OpenRAM/compiler/debug.py", line 33, in check
    assert 0
AssertionError
 -------------------------------------
OpenRAM Version :  v1.2.8
------------------------------------- CONFIG FILE -------------------------------------
# Number of bits per word
word_size = 8
#num_banks=1
# Number of 8-bit words (some people call this 'depth')
num_words = 32
# Read-Write ports
num_rw_ports = 1
# Read-only ports
num_r_ports = 0
# Write-only ports
num_w_ports = 0
num_spare_rows = 1
num_spare_cols = 1
# The fabrication technology. This must match the PDK name in $OPENRAM_TECH.
tech_name = "sky130"

# Process corners, temperature and voltage to characterize
#nomial_corner_only = True
use_conda = False
supply_voltages = [1.7, 1.8, 1.9]
temperatures = [25, 50, 100]
process_corners = ["SS", "TT", "FF"]

# Path and name for the output files
output_path = "temp"
output_name = "sram_{0}_{1}_{2}".format(word_size, num_words, tech_name)
------------------------------------- CONFIG FILE  END  -------------------------------------
[sram_8_32_sky130.log](https://github.com/VLSIDA/OpenRAM/files/11183373/sram_8_32_sky130.log)

samuelkcrow, 2023-04-11:
I am working on a fix for this

IslamElsadek97, 2023-04-26:
I encountered the same error, is there a specific way to solve it?

samuelkcrow, 2023-05-05:
This was an issue specific to our custom layout of sky130 single port memories that should be patched in arecent openram version (1.2.12, should work in newer versions too).
I was able to successfully run openram with your configuration.
Currently openram is using a recent version of magic+netgen that does not seem to pass LVS with sky130 memories. If you want to run drc/lvs on this memory, I'd recommend you try with magic version 8.3.211 and netgen version 1.5.195, that's what I used to get it to pass LVS. We are working on getting the LVS issue patched but you are free to create another issue for that.