VLSIDA/OpenRAM #191: ROM generation tutorial/references

SPD3, 2023-04-12:
**Describe the bug**
No tutorials for how to use the rom compiler.

**Version**
85db2043b27f842f8b927fe1a49deb42b0b05b40

**To Reproduce**
N/A

**Expected behavior**
Resources to exist within the docs/source directory showing how to generate read only memory, including a sample config file.

**Logs**
N/A

**Additional context**
N/A

mguthaus, 2023-11-28:
Hi,

No there aren't any tutorials but this would be a great contribution. The usage is very similar to the SRAMs and there are examples in the macros subdirectory.

Matt

mguthaus, 2026-10-13:
Closing as stale. This has been inactive for 2.9 years and there is no current reproduction against `dev`, so it is not actionable as it stands.

Since this was filed the docs gained [basic_rom_usage.md](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/basic_rom_usage.md), which covers most of what was asked for.

To be clear, this is being closed for inactivity rather than rejected. If it still affects you, please reopen or file a fresh issue with a reproduction against current `dev`, including your configuration file, technology, OS and `python3 --version`, and the contents of the temp directory OpenRAM leaves behind on failure ([the FAQ explains which files those are](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself, a PR would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch.

closed 2026-10-13 by mguthaus