VLSIDA/OpenRAM #193: make pdk broken at the moment? Error 127

jackpanderson, 2023-04-28:
**Describe the bug**
make pdk seems to no longer work. A colleague and I are both running into the same error trying to set up OpenRAM on a new machine. 

**Version**
Which commit are you using?

**To Reproduce**
Got to the point of installation where I ran "make build" in the docker directory, then tried "make pdk" in the OpenRAM directory

**Expected behavior**
Installing OpenRAM

**Logs**
jande180@lab-XPS-8950:~/CARP/OpenRAM$ make pdk
Cloning skywater PDK...
HEAD is now at f70d8ca Merge pull request #346 from QuantamHD/add_reram
Installing open_pdks...
/bin/sh: 1: source: not found
make: *** [Makefile:75: /home/jande180/CARP/OpenRAM/sky130A] Error 127

erendn, 2023-04-28:
This happened because we changed the Makefile to use conda when installing the PDK. You can disable it by removing [this](https://github.com/VLSIDA/OpenRAM/blob/stable/Makefile#L53) line.

erendn, 2023-04-29:
This issue should be fixed with [v1.2.11](https://github.com/VLSIDA/OpenRAM/releases/tag/v1.2.11)

closed 2023-04-29 by erendn