VLSIDA/OpenRAM #194: Make pdk still broken, having trouble installing with conda

jackpanderson, 2023-04-30:
**Describe the bug**
Trying to run the installation as detailed in the documentation. I raised an issue about this yesterday and was said to be solved, but I am having issues still with OpenRAM or Conda.

**Version**
Most recent

**To Reproduce**
Running make pdk after installing conda still is not working on my end.

**Expected behavior**
Installing sucessfully.

**Logs**
(OpenRAM) jande180@lab-XPS-8950:~/CARP/OpenRAM$ make pdk
Cloning skywater PDK...
HEAD is now at f70d8ca Merge pull request #346 from QuantamHD/add_reram
Installing open_pdks...
/bin/sh: 1: source: not found
make: *** [Makefile:75: /home/jande180/CARP/OpenRAM/sky130A] Error 127

erendn, 2023-04-30:
If you have installed conda, `make pdk` will activate the conda environment and use the Magic tool installed there. The Makefile uses `source ./miniconda/bin/activate` command to activate it. Your `/bin/sh` can't find the `source` command. `/bin/sh` is usually a symbolic link to your system shell.

This command must be available in `/bin/bash`. Maybe you can switch to that one? What is your system shell? You can run `echo $SHELL` to see it.

jackpanderson, 2023-04-30:
My system shell is bash

erendn, 2023-04-30:
Can you try again after adding the following line somewhere at the top of the Makefile? Your `/bin/sh` may not be linked to `/bin/bash`.
```
SHELL := /bin/bash
```

jackpanderson, 2023-04-30:
The make pdk worked, as did make install, but still running into issues with using make in the macros folder.

~/CARP/OpenRAM/macros$ make 32by2048_1rw1r
make: *** No rule to make target '32by2048_1rw1r'.  Stop.

erendn, 2023-04-30:
The make target doesn't match any config name in the config folders in macros. For example, you can try:
```
make freepdk45_sram_1rw1r_32x2048_8.ok
```

jackpanderson, 2023-04-30:
That matches the name of one of my configs that I added to the sram_configs folder. Any reason for having to add the .ok at the end of the config name? Additionally, running into this issue where the path to the temp.gds is not found.

[openram.router.router_tech/__init__]: Minimum track wire width: 0.380
Traceback (most recent call last):
  File "/home/jande180/CARP/OpenRAM/sram_compiler.py", line 71, in <module>
    s = sram()
  File "/home/jande180/CARP/OpenRAM/compiler/sram.py", line 58, in __init__
    self.s.create_layout()
  File "/home/jande180/CARP/OpenRAM/compiler/modules/sram_1bank.py", line 216, in create_layout
    self.route_layout()
  File "/home/jande180/CARP/OpenRAM/compiler/modules/sram_1bank.py", line 1080, in route_layout
    init_bbox = self.get_bbox(side="ring",
  File "/home/jande180/CARP/OpenRAM/compiler/base/hierarchy_layout.py", line 1805, in get_bbox
    self.gds_write(gds_filename)
  File "/home/jande180/CARP/OpenRAM/compiler/base/hierarchy_layout.py", line 1492, in gds_write
    writer.writeToFile(gds_name)
  File "/home/jande180/CARP/OpenRAM/compiler/gdsMill/gdsMill/gds2writer.py", line 563, in writeToFile
    self.fileHandle = open(fileName,"wb")
FileNotFoundError: [Errno 2] No such file or directory: '/results/32by2048_1rw1r/tmp/temp.gds'
make: *** [Makefile:84: 32by2048_1rw1r.ok] Error 1

erendn, 2023-04-30:
We have `.ok` at the end because it's how the Makefile targets are defined [here](https://github.com/VLSIDA/OpenRAM/blob/stable/macros/Makefile#L79).

Can you make sure that this path exists? Notice that it is an absolute path, not a relative one. Maybe you wanted `./results/32by2048_1rw1r/tmp/temp.gds`?

mguthaus, 2023-04-30:
We used to not have the ok in the target but it got added somehow. I think we used to have an additional target without the .ok as well? It's supposed to create a "stamp" file with .ok just to know that it was run and not run again until deleted or cleaned.

If we can remove the .ok from the target that would be more intuitive.

jackpanderson, 2023-05-04:
On the issue of the path to the temp.gds, I am confused as to why I would specify a path to that. It seems like the flow is trying to open a temporary file it had previously created, but is looking in the wrong place. Running the flow with some of the example modules results in the same error.

erendn, 2023-05-04:
I think I found the root cause of this issue. Can you look at what OpenRAM's log says about "temp" directory. Seems like it fails to create that directory but doesn't give an error.

jackpanderson, 2023-05-04:
[openram.globals/init_openram]: Initializing OpenRAM...
[openram.globals/setup_paths]: OpenRAM source code found in /home/jande180/CARP/OpenRAM/compiler
[openram.globals/setup_paths]: Temporary files saved in /results/32by2048_1rw1r/tmp/
ERROR: file globals.py: line 328: Configuration file name is not a valid Python module name: 32by2048_1rw1r. It should be a valid identifier.

[openram.globals/read_config]: Configuration file is /home/jande180/CARP/OpenRAM/macros/sram_configs/32by2048_1rw1r.py
[openram.globals/read_config]: Output saved in /home/jande180/CARP/OpenRAM/macros/32by2048_1rw1r/
[openram.globals/install_conda]: Creating conda setup...
[openram.globals/import_tech]: Tech directory found in /home/jande180/CARP/OpenRAM/technology:/home/jande180/.local/lib/python3.10/site-packages/openram/technology
[openram.globals/import_tech]: Adding technology path: /home/jande180/CARP/OpenRAM/technology
[openram.globals/import_tech]: Adding technology path: /home/jande180/.local/lib/python3.10/site-packages/openram/technology
[openram.globals/init_paths]: Creating temp directory: /results/32by2048_1rw1r/tmp/
[openram.globals/setup_bitcell]: Using bitcell: bitcell_2port
[openram.characterizer/<module>]: Initializing characterizer...
[openram.characterizer/<module>]: Analytical model enabled.
[openram.verify/<module>]: Initializing verify...
[openram.verify/<module>]: LVS/DRC/PEX disabled.
[openram.globals/setup_bitcell]: Using bitcell: bitcell_2port
|==============================================================================|
|=========                      OpenRAM v1.2.11                       =========|
|=========                                                            =========|
|=========               VLSI Design and Automation Lab               =========|
|=========        Computer Science and Engineering Department         =========|
|=========            University of California Santa Cruz             =========|
|=========                                                            =========|
|=========          Usage help: openram-user-group@ucsc.edu           =========|
|=========        Development help: openram-dev-group@ucsc.edu        =========|
|=========                See LICENSE for license info                =========|
|==============================================================================|
** Start: 05/04/2023 15:49:22
Technology: sky130
Total size: 65536 bits
Word size: 32
Words: 2048
Banks: 1
RW ports: 1
R-only ports: 1
W-only ports: 0
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
Words per row: None
Output files are:
/home/jande180/CARP/OpenRAM/macros/32by2048_1rw1r/32by2048_1rw1r.lvs
/home/jande180/CARP/OpenRAM/macros/32by2048_1rw1r/32by2048_1rw1r.sp
/home/jande180/CARP/OpenRAM/macros/32by2048_1rw1r/32by2048_1rw1r.v
/home/jande180/CARP/OpenRAM/macros/32by2048_1rw1r/32by2048_1rw1r.lib
/home/jande180/CARP/OpenRAM/macros/32by2048_1rw1r/32by2048_1rw1r.py
/home/jande180/CARP/OpenRAM/macros/32by2048_1rw1r/32by2048_1rw1r.html
/home/jande180/CARP/OpenRAM/macros/32by2048_1rw1r/32by2048_1rw1r.log
/home/jande180/CARP/OpenRAM/macros/32by2048_1rw1r/32by2048_1rw1r.lef
/home/jande180/CARP/OpenRAM/macros/32by2048_1rw1r/32by2048_1rw1r.gds
[openram.sram_config/recompute_sizes]: Recomputing with words per row: 8
[openram.sram_config/recompute_sizes]: Rows: 256 Cols: 256
[openram.sram_config/recompute_sizes]: Row addr size: 8 Col addr size: 3 Bank addr size: 11
[openram.sram_config/compute_sizes]: Set SRAM Words Per Row=8
[openram.modules.bitcell_base_array/__init__]: Creating 32by2048_1rw1r_capped_replica_bitcell_array 256 x 256
[openram.modules.capped_replica_bitcell_array/__init__]: Creating 32by2048_1rw1r_capped_replica_bitcell_array 256 x 256 rbls: [1, 1] left_rbl: [0] right_rbl: [1]
[openram.modules.bitcell_base_array/__init__]: Creating 32by2048_1rw1r_replica_bitcell_array 256 x 256
[openram.modules.replica_bitcell_array/__init__]: Creating 32by2048_1rw1r_replica_bitcell_array 256 x 256 rbls: [1, 1] left_rbl: [0] right_rbl: [1]
[openram.modules.bitcell_base_array/__init__]: Creating 32by2048_1rw1r_bitcell_array 256 x 256
[openram.modules.bitcell_array/__init__]: Creating 32by2048_1rw1r_bitcell_array 256 x 256
[openram.modules.bitcell_base_array/__init__]: Creating 32by2048_1rw1r_replica_column 258 x 1
[openram.modules.bitcell_base_array/__init__]: Creating 32by2048_1rw1r_replica_column_0 258 x 1
[openram.modules.bitcell_base_array/__init__]: Creating 32by2048_1rw1r_dummy_array 1 x 256
[openram.modules.bitcell_base_array/__init__]: Creating 32by2048_1rw1r_col_cap_array 1 x 258
[openram.modules.bitcell_base_array/__init__]: Creating 32by2048_1rw1r_col_cap_array_0 1 x 258
[openram.modules.bitcell_base_array/__init__]: Creating 32by2048_1rw1r_row_cap_array 258 x 1
[openram.modules.bitcell_base_array/__init__]: Creating 32by2048_1rw1r_row_cap_array_0 258 x 1
[openram.modules.and2_dec/__init__]: Creating and2_dec and2_dec
[... 16 lines cut ...]
[openram.modules.control_logic_base/__init__]: Creating control_logic_rw
[openram.modules.dff_buf/__init__]: Creating 32by2048_1rw1r_dff_buf
[openram.modules.dff_buf_array/__init__]: Creating 32by2048_1rw1r_dff_buf_array
[openram.modules.dff_buf/__init__]: Creating 32by2048_1rw1r_dff_buf_0
[... cut ...]
[openram.modules.dff_buf_array/__init__]: Creating 32by2048_1rw1r_dff_buf_array_0
[openram.modules.pdriver/__init__]: creating pdriver pdriver_6
** Submodules: 25.6 seconds
** Placement: 0.0 seconds
[openram.router.router_tech/__init__]: Minimum track width: 0.680
[openram.router.router_tech/__init__]: Minimum track space: 0.300
[openram.router.router_tech/__init__]: Minimum track wire width: 0.380

This is the entire log from the failed run. One of the first lines seems to try to make the results directory but fails.

erendn, 2023-05-04:
I'll try to fix this issue. For now, you can define `openram_temp` in your config file to a path that actually exists so that OpenRAM can write the temporary files.

jackpanderson, 2023-05-04:
Put that into my python file in the sram_configs file? I've tried your suggestion and it seems like adding this parameter does not make a difference where it tries to create the temporary files.

jackpanderson, 2023-05-04:
Update, it was likely an issue with permissions, as I am running the OpenRAM on a machine at my university which has different permissions. I added sudo and ran and it seems to not be failing out and has the ability to make the directory.

closed 2023-05-04 by jackpanderson

erendn, 2023-05-05:
OpenRAM should now throw an error if it can't make the temp directory (420ce01b46d2db06e0132e5f18f0e0dcd86949ba, [v1.2.13](https://github.com/VLSIDA/OpenRAM/releases/tag/v1.2.13)).