VLSIDA/OpenRAM #196: Adding linear regression model breaks flow

jackpanderson, 2023-05-13:
**Describe the bug**
As said in the title, adding the linear regression model in the .py config settings breaks the characterization. 

**Version**
1.2.15

**To Reproduce**
Adding model_name = "linear_regression" to the model config

**Logs**
LIB: Characterizing...
[openram.characterizer.lib/prepare_tables]: Slews: [0.00125 0.005   0.04   ]
[openram.characterizer.lib/prepare_tables]: Loads: [ 1.7225  6.89   27.56  ]
[openram.characterizer.lib/prepare_tables]: self.load_slews : [(1.7225, 0.00125), (6.89, 0.00125), (27.56, 0.00125), (1.7225, 0.005), (6.89, 0.005), (27.56, 0.005), (1.7225, 0.04), (6.89, 0.04), (27.56, 0.04)]
[openram.characterizer.lib/characterize_corners]: Characterizing corners: [('TT', 1.8, 25)]
[openram.characterizer.lib/characterize_corners]: Corner: ('TT', 1.8, 25)
[openram.characterizer.lib/characterize_corners]: Writing to /home/jande180/CARP/OpenRAM/macros/sram_32_32_sky130_1r1rw/sram_32_32_sky130_1r1rw_TT_1p8V_25C.lib
[openram.characterizer.regression_model/get_lib_values]: Characterizing SRAM using regression models.
Traceback (most recent call last):
  File "/home/jande180/CARP/OpenRAM/sram_compiler.py", line 74, in <module>
    s.save()
  File "/home/jande180/CARP/OpenRAM/compiler/sram.py", line 163, in save
    lib(out_dir=OPTS.output_path, sram=self.s, sp_file=sp_file)
  File "/home/jande180/CARP/OpenRAM/compiler/characterizer/lib.py", line 46, in __init__
    self.characterize_corners()
  File "/home/jande180/CARP/OpenRAM/compiler/characterizer/lib.py", line 163, in characterize_corners
    self.characterize()
  File "/home/jande180/CARP/OpenRAM/compiler/characterizer/lib.py", line 175, in characterize
    self.compute_delay()
  File "/home/jande180/CARP/OpenRAM/compiler/characterizer/lib.py", line 646, in compute_delay
    char_results = m.get_lib_values(self.load_slews)
  File "/home/jande180/CARP/OpenRAM/compiler/characterizer/regression_model.py", line 72, in get_lib_values
    models = self.train_models()
  File "/home/jande180/CARP/OpenRAM/compiler/characterizer/regression_model.py", line 138, in train_models
    self.output_names = get_data_names(data_path)[self.num_inputs:]
  File "/home/jande180/CARP/OpenRAM/compiler/characterizer/analytical_util.py", line 22, in get_data_names
    with open(file_name, newline='') as csvfile:
FileNotFoundError: [Errno 2] No such file or directory: '/home/jande180/CARP/OpenRAM/technology/sky130/sim_data/sim_data.csv'
make: *** [Makefile:82: sram_32_32_sky130_1r1rw.ok] Error 1

mguthaus, 2023-05-14:
This mode requires you to precharacterize a bunch of memories in that file so that it can use regression to predict models for new memories. We haven't done that for Skywater yet and instead recommend regular characterization. This is from the following MS thesis:
https://www.proquest.com/openview/273a1089a88606160de418f5b75f29e7/1.pdf?pq-origsite=gscholar&cbl=18750&diss=y

The error should definitely be improved though.

mguthaus, 2026-10-13:
Closing as stale. This has been inactive for 3.4 years and there is no current reproduction against `dev`, so it is not actionable as it stands.

There is now a `23_lib_sram_linear_regression_test` in the regression suite, so the flow is at least exercised.

To be clear, this is being closed for inactivity rather than rejected. If it still affects you, please reopen or file a fresh issue with a reproduction against current `dev`, including your configuration file, technology, OS and `python3 --version`, and the contents of the temp directory OpenRAM leaves behind on failure ([the FAQ explains which files those are](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself, a PR would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch.

closed 2026-10-13 by mguthaus