VLSIDA/OpenRAM #197: the meaning of local_array_size

hanm2019, 2023-05-29:
I found that there is a parameter named `local_array_size` in freepdk45_sram_1r21r_32x2048_8.py, but I can not find any document which describes  this parameter. 
 https://github.com/VLSIDA/OpenRAM/blob/0a4a36488b21137d41755f8d0540b747f625b8bd/macros/sram_configs/freepdk45_sram_1rw1r_32x2048_8.py#L5

mguthaus, 2023-05-29:
This parameter specifies the size of a local/sub array and uses a hierarchical word line. I don't believe it was working yet in sky130 due to a DRC error.

closed 2023-05-29 by mguthaus

hanm2019, 2023-05-29:
How should I config this parameter in FPDK45nm? set the value equal to the `word_size`?

mguthaus, 2023-05-29:
I suggest you look at how it affects layout. Setting it to the word size is the default and will not use a hierarchical word line. Larger arrays can have a big word line delay and can benefit from this to improve speed.