VLSIDA/OpenRAM #198: How to use the .sp&.v generated after executing config on other tools

haotommy, 2023-07-03:
I am running the generated .sp file using Hspice, but there is no circuit. I cannot find where to modify the library in the .sp file, or which tool I can use to view my .sp file?

The .v file refers to the generated memory content and does not include the surrounding circuitry. It focuses solely on the memory itself. Regarding the absence of reset and the presence of T_Hold in the file, if you want to modify them for waveform simulation using ModelSim, you would need to make the necessary changes manually?

Alternatively, could you please tell me which file controls the generation content of the .sp and .v files?

mguthaus, 2023-07-03:
I'm not sure what you mean by "no circuit". The sp file is the circuit and can be used for either simulation or LVS.

The Verilog file is just for behavioral modeling and is not synthesizable. The memory is for use by your own project and doesn't have any surrounding circuitry.

closed 2023-07-03 by mguthaus

haotommy, 2023-07-05:
[image: image.png]
This is the result I obtained when running the generated .sp file with
Hspice: It shows an error message: "no element found in circuit!!" However,
I cannot find where I can modify the process of my library.

[image: image.png]
The following files are the generated .v file for a 64x64 memory (with
added reset) and a self-written test bench.
When running ModelSim, I observe a red line in the waveform at dout0. Is
this due to the issue with T_Hold?
Is the waveform generated from the .v file considered normal?

mguthaus, 2023-07-05:
Are you running spice on only the or are you using a stimulus file like the
functional one that is generated? I don't know what you mean by "modify the
process of my library".

Outputs are always invalid until the read completes so you can expect some
red.

Matt

haotommy, 2023-07-06:
Because I want to see the results of my .sp file, I would like to ask what
is the process node used for fabrication?
Specifically, what are the process nodes for freepdk45, scn3me_subm, and
scn4m_subm?

mguthaus, 2023-07-06:
https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/index.md#supported-technologies

haotommy, 2023-07-13:
Since I want to run .sp simulations, does OpenRAM provide process files?
Or is the technology file itself the process file?

haotommy, 2023-07-14:
> These are provided in the PDK.

The PDK about these process is in the any files with OpenRAM?

haotommy, 2023-07-18:
Do I have to contact them to provide the PDK, or does OpenRAM itself have
it?

mguthaus, 2023-07-18:
You install the PDK like this:

https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/basic_setup.md#sky130-setup

The other PDKs are included.

Matt

haotommy, 2023-07-20:
So I also need to use the OpenRAM to  find the PDK about these process?
Can I directly modify the .sp to change the circuit's content to another
fabricated circuit?