VLSIDA/OpenRAM #199: data not getting written into memory cells

KolliparaAbhishekAnand, 2023-07-15:
**Describe the bug**
In memory macros of size 16_64 , 16_128 , data is not getting written into memory cells . this is observed  by testing on generated .sp files using ngspice as simulator tool. 

**Version**
v1.2.18 and commit 3620d56790a7153f20a509a6e804af518cde763b

**To Reproduce**
1.I wrote different content into memory cells , and every time the data stored is always zero in memory cells. 
2. also tested the functional_stim.sp file generated by openram . the file will include some pre-known results which we should be able to get by running this test stimulus file . pre-known results and actual simulation results didn't match.`

**Expected behavior**
data should be written into memory cells properly. i observed this behaviour in those memory macros with word size 64 and 128.

bugraonal, 2023-07-19:
Can you attach the config you used. I am having trouble replicating this issue.

KolliparaAbhishekAnand, 2023-07-19:
# Data word size
word_size = 64
# Number of words in the memory
num_words = 16

human_byte_size = "{:.0f}kbytes".format((word_size * num_words)/1024/8)

netlist_only=True

num_spare_cols=1
num_spare_rows=1
tech_name = "sky130"
process_corners = [ "TT"]
supply_voltages = [ 1.8 ]
temperatures = [ 27 ]
num_rw_ports=1
num_r_ports=0
num_w_ports=0
write_size=None
ports_human='1rw'

# Disable analytical models for full characterization (WARNING: slow!)
# analytical_delay = False

file_path="/home/abhishek/SINESyslab/OpenRAM/macros/sram_configs/"
import os
exec(open(os.path.join(os.path.dirname(file_path),'sky130_sram_common.py')).read())

bugraonal, 2023-07-19:
I am able to read and write with these stimulus files.
[functional.tar.gz](https://github.com/VLSIDA/OpenRAM/files/12099701/functional.tar.gz)

Could you perhaps send your stimulus?

KolliparaAbhishekAnand, 2023-07-19:
I am getting problem in those circuits in which number of words are 16 and data bits/word size = 64 . you have taken exactly opposite of it.

bugraonal, 2023-07-19:
Ah, I see. I'll try that one.

KolliparaAbhishekAnand, 2023-07-20:
are you able to replicate the issue??