VLSIDA/OpenRAM #200: having issue while trying to create sram for sky130 pdk

bharath19-gs, 2023-07-15:
Hi,

I am facing issue with sky130, not getting the required output as per requirement
please find the log below and the config that i have used.

|==============================================================================|
|=========                      OpenRAM v1.2.19                       =========|
|=========                                                            =========|
|=========               VLSI Design and Automation Lab               =========|
|=========        Computer Science and Engineering Department         =========|
|=========            University of California Santa Cruz             =========|
|=========                                                            =========|
|=========          Usage help: openram-user-group@ucsc.edu           =========|
|=========        Development help: openram-dev-group@ucsc.edu        =========|
|=========                See LICENSE for license info                =========|
|==============================================================================|
** Start: 07/15/2023 15:51:44
Technology: sky130
Total size: 4096 bits
Word size: 16
Words: 256
Banks: 1
RW ports: 1
R-only ports: 0
W-only ports: 0
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
Words per row: None
Output files are: 
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.lvs
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.sp
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.v
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.lib
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.py
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.html
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.log
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.lef
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.gds
ERROR: file hierarchy_layout.py: line 579: Should use a pin iterator since more than one pin vpb

ERROR: file hierarchy_layout.py: line 586: No pin found with name vpb on sky130_fd_bd_sram__sram_sp_colenda. Saved as missing_pin.gds.

my config 
# Data word size
word_size = 16
# Number of words in the memory
num_words = 256

num_spare_cols=1
num_spare_rows=1

num_rw_ports = 1

# Technology to use in $OPENRAM_TECH
tech_name = "sky130"
# Process corners to characterize
process_corners = [ "TT" ]
# Voltage corners to characterize
supply_voltages = [ 3.3 ]
# Temperature corners to characterize
temperatures = [ 25 ]

# Output directory for the results
output_path = "temp"
# Output file base name
output_name = "fakeram130_256x16"

# Disable analytical models for full characterization (WARNING: slow!)
# analytical_delay = False

# To force this to use magic and netgen for DRC/LVS/PEX
# Could be calibre for FreePDK45
drc_name = "magic"
lvs_name = "netgen"
pex_name = "magic"

and 
using the command :
python3 $OPENRAM_HOME/../sram_compiler.py myconfig.py

the requirement: want to create a sram of 256x16 

please do let me know how to fix this.

regards 
gsb

bharath19-gs, 2023-07-15:
** Start: 07/15/2023 20:15:27
Technology: sky130
Total size: 4096 bits
Word size: 16
Words: 256
Banks: 1
RW ports: 1
R-only ports: 0
W-only ports: 0
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
Words per row: None
Output files are: 
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.lvs
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.sp
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.v
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.lib
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.py
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.html
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.log
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.lef
/home/gsb/Documents/OpenRAM/compiler/temp/fakeram130_256x16.gds
ERROR: file hierarchy_layout.py: line 579: Should use a pin iterator since more than one pin vpb
ERROR: file hierarchy_layout.py: line 586: No pin found with name vpb on sky130_fd_bd_sram__sram_sp_colenda. Saved as missing_pin.gds.
Traceback (most recent call last):
  File "/home/gsb/Documents/OpenRAM/compiler/base/hierarchy_layout.py", line 579, in get_pin
    debug.error("Should use a pin iterator since more than one pin {}".format(text), -1)
  File "/home/gsb/Documents/OpenRAM/compiler/debug.py", line 47, in error
    assert return_value == 0
           ^^^^^^^^^^^^^^^^^
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/gsb/Documents/OpenRAM/compiler/../sram_compiler.py", line 71, in <module>
    s = sram()
        ^^^^^^
  File "/home/gsb/Documents/OpenRAM/compiler/sram.py", line 56, in __init__
    self.s.create_netlist()
  File "/home/gsb/Documents/OpenRAM/compiler/modules/sram_1bank.py", line 196, in create_netlist
    self.add_modules()
  File "/home/gsb/Documents/OpenRAM/compiler/modules/sram_1bank.py", line 462, in add_modules
    self.bank = factory.create("bank", sram_config=self.sram_config, module_name="bank")
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gsb/Documents/OpenRAM/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gsb/Documents/OpenRAM/compiler/modules/bank.py", line 49, in __init__
    self.create_netlist()
  File "/home/gsb/Documents/OpenRAM/compiler/modules/bank.py", line 59, in create_netlist
    self.add_modules()
  File "/home/gsb/Documents/OpenRAM/compiler/modules/bank.py", line 377, in add_modules
    self.bitcell_array = factory.create(module_type="capped_replica_bitcell_array",
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gsb/Documents/OpenRAM/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gsb/Documents/OpenRAM/technology/sky130/custom/sky130_capped_replica_bitcell_array.py", line 37, in __init__
    self.create_netlist()
  File "/home/gsb/Documents/OpenRAM/technology/sky130/custom/sky130_capped_replica_bitcell_array.py", line 43, in create_netlist
    self.add_modules()
  File "/home/gsb/Documents/OpenRAM/technology/sky130/custom/sky130_capped_replica_bitcell_array.py", line 48, in add_modules
    self.replica_bitcell_array = factory.create(module_type="replica_bitcell_array",
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gsb/Documents/OpenRAM/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gsb/Documents/OpenRAM/technology/sky130/custom/sky130_replica_bitcell_array.py", line 61, in __init__
    super().__init__(self.row_size, self.column_size, rbl, left_rbl, right_rbl, name)
  File "/home/gsb/Documents/OpenRAM/technology/sky130/custom/replica_bitcell_array.py", line 72, in __init__
    self.create_netlist()
  File "/home/gsb/Documents/OpenRAM/technology/sky130/custom/replica_bitcell_array.py", line 82, in create_netlist
    self.add_modules()
  File "/home/gsb/Documents/OpenRAM/technology/sky130/custom/replica_bitcell_array.py", line 136, in add_modules
    self.replica_columns[port] = factory.create(module_type="replica_column",
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gsb/Documents/OpenRAM/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gsb/Documents/OpenRAM/technology/sky130/custom/sky130_replica_column.py", line 63, in __init__
    self.create_layout()
  File "/home/gsb/Documents/OpenRAM/technology/sky130/custom/sky130_replica_column.py", line 76, in create_layout
    self.add_layout_pins()
  File "/home/gsb/Documents/OpenRAM/technology/sky130/custom/sky130_replica_column.py", line 240, in add_layout_pins
    pin = inst.get_pin("vpb")
          ^^^^^^^^^^^^^^^^^^^
  File "/home/gsb/Documents/OpenRAM/compiler/base/geometry.py", line 253, in get_pin
    pin = copy.deepcopy(self.mod.get_pin(name))
                        ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gsb/Documents/OpenRAM/compiler/base/hierarchy_layout.py", line 586, in get_pin
    debug.error("No pin found with name {0} on {1}. Saved as missing_pin.gds.".format(name, self.cell_name), -1)
  File "/home/gsb/Documents/OpenRAM/compiler/debug.py", line 47, in error
    assert return_value == 0
           ^^^^^^^^^^^^^^^^^
AssertionError

extra details

hope it helps to find where i am going wrong or whats causing the issues

NamrathaCTR, 2023-08-24:
Hi, I am facing the same issue? @bharath19-gs , were you able to find a way to run this?

jcirimel, 2023-08-24:
Sorry I'm in the middle of a big refactor of this code right now. I'll have this bug fixed by the end of the week, probably within 24 hours.

jcirimel, 2023-08-26:
Should be fixed now

closed 2023-08-26 by jcirimel

NamrathaCTR, 2023-08-30:
Hi, for below configuration file(no of ports=1), I am getting the above mentioned same issue, but if I change no of ports to 2, I am not getting issue. I guess this issue for generating single port memories. Can you please go through my configuration file and tell me if I am making any mistake?

# Data word size
word_size = 8
# Number of words in the memory
num_words = 16

num_rw_ports = 1
num_r_ports = 0
num_w_ports = 0
num_spare_cols = 1
num_spare_rows = 1

# Technology to use in $OPENRAM_TECH
tech_name = "sky130"
# Process corners to characterize
process_corners = [ "TT" ]
# Voltage corners to characterize
supply_voltages = [ 3.3 ]
# Temperature corners to characterize
temperatures = [ 25 ]

# Output directory for the results
output_path = "temp"
# Output file base name
output_name = "sram_16x8"

# Disable analytical models for full characterization (WARNING: slow!)
# analytical_delay = False

# To force this to use magic and netgen for DRC/LVS/PEX
# Could be calibre for FreePDK45
drc_name = "magic"
lvs_name = "netgen"
pex_name = "magic"