VLSIDA/OpenRAM #201: Anaconda install problem

haotommy, 2023-07-21:
tommy@tommy-virtual-machine:~$ sudo apt install ngspice
tommy@tommy-virtual-machine:~$ sudo apt-get install python3
tommy@tommy-virtual-machine:~$ sudo apt install python3-pip
tommy@tommy-virtual-machine:~$ pip3 install numpy
tommy@tommy-virtual-machine:~$ pip3 install coverage
tommy@tommy-virtual-machine:~$ pip3 install scikit-learn
tommy@tommy-virtual-machine:~$ sudo apt update
tommy@tommy-virtual-machine:~$ sudo apt install git
tommy@tommy-virtual-machine:~$ sudo apt-get update
tommy@tommy-virtual-machine:~$ sudo apt-get install make
tommy@tommy-virtual-machine:~$ vim .bashrc
(Below are the environment variables I added.
export OPENRAM_HOME="$HOME/OpenRAM/compiler"
export OPENRAM_TECH="$HOME/OpenRAM/technology"
export PYTHONPATH="$PYTHONPATH:$OPENRAM_HOME"
export PYTHONPATH="$OPENRAM_HOME:$OPENRAM_TECH/sky130:$OPENRAM_TECH/sky130/custom"
export CONDA_HOME="/path/to/conda/setup"
After adding them, close the Terminal and then reopen it.)
tommy@tommy-virtual-machine:~$pip3 install openram
tommy@tommy-virtual-machine:~$pip3 install git+https://git@github.com/VLSIDA/OpenRAM.git@dev
tommy@tommy-virtual-machine:~$git clone https://github.com/VLSIDA/OpenRAM.git
tommy@tommy-virtual-machine:~$cd OpenRAM
tommy@tommy-virtual-machine:~/OpenRAM$ sudo pip install -r requirements.txt
tommy@tommy-virtual-machine:~/OpenRAM$./install_conda.sh
It can't find the Anaconda.

mguthaus, 2023-07-21:
The script itself installs miniconda.

You can see installation instructions at:
https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/basic_setup.md#basic-setup

I believe I shared this with you before, so please don't do all that other stuff...

closed 2023-07-21 by mguthaus

haotommy, 2023-07-21:
Hello,
I had successfully installed OpenRAM before, but due to the virtual server
failure, I need to reinstall it.
However, following the steps, I encountered some issues.
***************************************************************************************************************
***@***.***:~$ sudo apt install ngspice
***@***.***:~$ sudo apt-get install python3
***@***.***:~$ sudo apt install python3-pip
***@***.***:~$ pip3 install numpy
***@***.***:~$ pip3 install coverage
***@***.***:~$ pip3 install scikit-learn
***@***.***:~$ sudo apt update
***@***.***:~$ sudo apt install git
***@***.***:~$ sudo apt-get update
***@***.***:~$ sudo apt-get install make
***@***.***:~$ vim .bashrc
[image: image.png]
(Below are the environment variables I added.
export OPENRAM_HOME="$HOME/OpenRAM/compiler"
export OPENRAM_TECH="$HOME/OpenRAM/technology"
export PYTHONPATH="$PYTHONPATH:$OPENRAM_HOME"
export
PYTHONPATH="$OPENRAM_HOME:$OPENRAM_TECH/sky130:$OPENRAM_TECH/sky130/custom"
export CONDA_HOME="/path/to/conda/setup"
After adding them, close the Terminal and then reopen it.)
***@***.***:~$pip3 install openram
***@***.***:~$pip3 install git+
***@***.******@***.***
***@***.***:~$git clone
https://github.com/VLSIDA/OpenRAM.git
***@***.***:~$cd OpenRAM
***@***.***:~/OpenRAM$./install_conda.sh
[image: image.png]
***********************************************************************************************************
I follow the steps on github but it gives me the wrong state.
Is there any mistake or missing step in the installation?
Thaks for your help.

mguthaus, 2023-07-21:
This isn't a real path:

export CONDA_HOME="/path/to/conda/setup"

mguthaus, 2023-07-21:
All python requirements are in requirements.txt

mguthaus, 2023-07-21:
Conda will install necessary tools for you automatically when you run install_conda.sh.

Everything else isn't needed.

haotommy, 2023-07-21:
I rewrite the  .bashrc with this export
CONDA_HOME="$HOME/path/to/conda/setup".
And then,I have two questions.
1.
***@***.***:~$ cd OpenRAM/
***@***.***:~/OpenRAM$ ./install_conda.sh
After it finish the process,it's that right away to have do this command
" ./install_conda.sh" in OpenRAM file?
***************************************************************************************************************************
2.
Why I doing Sky130 setup get problem?
***@***.***:~/OpenRAM$ make pdk
[image: image.png]
[image: image.png]
[image: image.png]
And I,
***@***.***:~/OpenRAM$ ./install_conda.sh
***@***.***:~/OpenRAM$ source miniconda/bin/activate
[image: image.png]
Finally,
***@***.***:~/OpenRAM$ make install
[image: image.png]
There have three problem in Sky130 Setup.

haotommy, 2023-07-22:
I rewrite the  .bashrc with this export
CONDA_HOME="$HOME/path/to/conda/setup".
It's that correct?
And then,I have two questions.
1.
***@***.***:~$ cd OpenRAM/
***@***.***:~/OpenRAM$ ./install_conda.sh
After it finish the process,it's that right away to have do this command
" ./install_conda.sh" in OpenRAM file?
***************************************************************************************************************************
2.
Why I doing Sky130 setup get problem?
***@***.***:~/OpenRAM$ make pdk
[image: image.png]
[image: image.png]
[image: image.png]
And I,
***@***.***:~/OpenRAM$ ./install_conda.sh
***@***.***:~/OpenRAM$ source miniconda/bin/activate
[image: image.png]
Finally,
***@***.***:~/OpenRAM$ make install
[image: image.png]
There have three problem in Sky130 Setup.

haotommy, 2023-07-22:
I have been looking at GitHub, but I am having trouble understanding it.
Could you please record a video tutorial for installing OpenRAM and running
simulations with .sp files?
I'm having trouble understanding the process from GitHub, no matter how
much I try to follow the instructions.