VLSIDA/OpenRAM #202: installaion problem and configuratiom problem

haotommy, 2023-07-25:
tommy@tommy-virtual-machine:~$ sudo apt install ngspice
tommy@tommy-virtual-machine:~$ sudo apt-get install python3
tommy@tommy-virtual-machine:~$ sudo apt install python3-pip
tommy@tommy-virtual-machine:~$ pip3 install numpy
tommy@tommy-virtual-machine:~$ pip3 install coverage
tommy@tommy-virtual-machine:~$ pip3 install scikit-learn
tommy@tommy-virtual-machine:~$ sudo apt update
tommy@tommy-virtual-machine:~$ sudo apt install git
tommy@tommy-virtual-machine:~$ sudo apt-get update
tommy@tommy-virtual-machine:~$ sudo apt-get install make
tommy@tommy-virtual-machine:~$ vim .bashrc
(Below are the environment variables I added.
export OPENRAM_HOME="$HOME/OpenRAM/compiler"
export OPENRAM_TECH="$HOME/OpenRAM/technology"
export PYTHONPATH="$PYTHONPATH:$OPENRAM_HOME"
export PYTHONPATH="$OPENRAM_HOME:$OPENRAM_TECH/sky130:$OPENRAM_TECH/sky130/custom"
export CONDA_HOME="$HOME/path/to/conda/setup")
It's the CONDA_HOME bash right?
***************************************************************************************************************** 
tommy@tommy-virtual-machine:~/OpenRAM$ ./install_conda.sh
It will get
./install_conda.sh: line 33: conda: command not found
******************************************************************************************************************
myconfig:
# Data word size
word_size = 64
# Number of words in the memory
num_words = 64

# Technology to use in $OPENRAM_TECH
tech_name = "scn4m_subm"

# You can use the technology nominal corner only
nominal_corner_only = True
# Or you can specify particular corners
# Process corners to characterize
# process_corners = ["SS", "TT", "FF"]
# Voltage corners to characterize
# supply_voltages = [ 3.0, 3.3, 3.5 ]
# Temperature corners to characterize
# temperatures = [ 0, 25, 100]

# Output directory for the results
output_path = "64*64"
# Output file base name
output_name = "sram_{0}_{1}_{2}".format(word_size,num_words,tech_name)

# Disable analytical models for full characterization (WARNING: slow!)
# analytical_delay = False

Go back to terminal:
tommy@tommy-virtual-machine:~$ python3 $OPENRAM_HOME/../sram_compiler.py myconfig
ERROR: file globals.py: line 326: Unable to read configuration file: /home/tommy/myconfig
Traceback (most recent call last):
  File "/home/tommy/OpenRAM/compiler/globals.py", line 324, in read_config
    config = importlib.import_module(module_name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'myconfig'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/tommy/OpenRAM/compiler/../sram_compiler.py", line 42, in <module>
    openram.init_openram(config_file=args[0])
  File "/home/tommy/OpenRAM/compiler/globals.py", line 186, in init_openram
    read_config(config_file, is_unit_test)
  File "/home/tommy/OpenRAM/compiler/globals.py", line 326, in read_config
    debug.error("Unable to read configuration file: {0}".format(config_file), 2)
  File "/home/tommy/OpenRAM/compiler/debug.py", line 47, in error
    assert return_value == 0
AssertionError

Why it error?
Or I enter the command in wrong file?
***********************************************************************************************************
I have been looking at GitHub, but I am having trouble understanding it.
Could you please record a video tutorial for installing OpenRAM and running simulations with .sp files? 
I'm having trouble understanding the process from GitHub, no matter how much I try to follow the instructions.

mguthaus, 2023-07-25:
There seems to be an error installing conda on your system since it can't find conda.

closed 2023-07-25 by mguthaus

haotommy, 2023-07-25:
Where is the error in my steps about installing conda?

mguthaus, 2023-07-25:
I can't tell from what you have shown. Are there any messages?

mguthaus, 2023-07-25:
Don't set CONDA_HOME unless you need to locate it somewhere else. That path probably doesn't exist?

erendn, 2023-07-25:
You don't need to set `CONDA_HOME`. It's an optional variable if you have multiple conda environments for OpenRAM, or want to install conda somewhere outside the OpenRAM directory.

haotommy, 2023-07-25:
I rewrite the  .bashrc with this export
CONDA_HOME="$HOME/path/to/conda/setup".
It's that correct?
Or I don't need to  set CONDA_HOME in my .bashrc?
And then,I have two questions.
1.
***@***.***:~$ cd OpenRAM/
***@***.***:~/OpenRAM$ ./install_conda.sh
After it finish the process,it's that right away to have do this command
" ./install_conda.sh" in OpenRAM file?
***************************************************************************************************************************
2.
Why I doing Sky130 setup get problem?
***@***.***:~/OpenRAM$ make pdk
[image: image.png]
[image: image.png]
[image: image.png]
And I,
***@***.***:~/OpenRAM$ ./install_conda.sh
***@***.***:~/OpenRAM$ source miniconda/bin/activate
[image: image.png]
Finally,
***@***.***:~/OpenRAM$ make install
[image: image.png]
There have three problem in Sky130 Setup.

haotommy, 2023-07-25:
> Don't set CONDA_HOME. Also, that is probably an invalid path.

So I don't need to set CONDA_HOME in my .bashrc.
Where the command "./install_conda.sh" enter the path?
Is ***@***.***:~/OpenRAM$ ./install_conda.sh?
Or ***@***.***:~$ ./install_conda.sh?

mguthaus, 2023-07-25:
In the current directory under miniconda:

https://github.com/VLSIDA/OpenRAM/blob/ecd486e050b6243e747bd2e259b53ea768ddcce0/install_conda.sh#L4

haotommy, 2023-07-26:
I delete the CONDA_HOME in my .bashrc and I open the terminal.
(my python version is 3.10 and I also install the pip3.
And I use the pip3 to install the  numpy,coverage,scikit-learn)

***@***.***:~/OpenRAM$ pip3 install -r requirements.txt
Defaulting to user installation because normal site-packages is not
writeable
Requirement already satisfied: scikit-learn>=0.22.2 in
/home/tommy/.local/lib/python3.10/site-packages (from -r requirements.txt
(line 1)) (1.3.0)
Requirement already satisfied: coverage>=4.5.2 in
/home/tommy/.local/lib/python3.10/site-packages (from -r requirements.txt
(line 2)) (7.2.7)
Requirement already satisfied: scipy>=1.3.3 in
/home/tommy/.local/lib/python3.10/site-packages (from -r requirements.txt
(line 3)) (1.11.1)
Requirement already satisfied: numpy>=1.17.4 in
/home/tommy/.local/lib/python3.10/site-packages (from -r requirements.txt
(line 4)) (1.25.1)
Requirement already satisfied: python-subunit>=1.4.0 in
/home/tommy/.local/lib/python3.10/site-packages (from -r requirements.txt
(line 5)) (1.4.2)
Requirement already satisfied: unittest2>=1.1.0 in
/home/tommy/.local/lib/python3.10/site-packages (from -r requirements.txt
(line 6)) (1.1.0)
Requirement already satisfied: joblib>=1.1.1 in
/home/tommy/.local/lib/python3.10/site-packages (from
scikit-learn>=0.22.2->-r requirements.txt (line 1)) (1.3.1)
Requirement already satisfied: threadpoolctl>=2.0.0 in
/home/tommy/.local/lib/python3.10/site-packages (from
scikit-learn>=0.22.2->-r requirements.txt (line 1)) (3.2.0)
Requirement already satisfied: extras in
/home/tommy/.local/lib/python3.10/site-packages (from
python-subunit>=1.4.0->-r requirements.txt (line 5)) (1.0.0)
Requirement already satisfied: testtools>=0.9.34 in
/home/tommy/.local/lib/python3.10/site-packages (from
python-subunit>=1.4.0->-r requirements.txt (line 5)) (2.6.0)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
at 0x7f9a741f0a90>: Failed to establish a new connection: [Errno -3]
Temporary failure in name resolution')': /simple/argparse/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
at 0x7f9a741f15d0>: Failed to establish a new connection: [Errno -3]
Temporary failure in name resolution')': /simple/argparse/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
at 0x7f9a741f16f0>: Failed to establish a new connection: [Errno -3]
Temporary failure in name resolution')': /simple/argparse/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
at 0x7f9a741f2a10>: Failed to establish a new connection: [Errno -3]
Temporary failure in name resolution')': /simple/argparse/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
at 0x7f9a741f2bc0>: Failed to establish a new connection: [Errno -3]
Temporary failure in name resolution')': /simple/argparse/
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
at 0x7f9a74242cb0>: Failed to establish a new connection: [Errno -3]
Temporary failure in name resolution')': /simple/unittest2/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
[... 376 lines cut ...]
at 0x7f68a7cab670>: Failed to establish a new connection: [Errno -3]
Temporary failure in name resolution')': /simple/scikit-learn/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
at 0x7f68a7caadd0>: Failed to establish a new connection: [Errno -3]
Temporary failure in name resolution')': /simple/scikit-learn/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
at 0x7f68a7caac50>: Failed to establish a new connection: [Errno -3]
Temporary failure in name resolution')': /simple/scikit-learn/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
at 0x7f68a7cab310>: Failed to establish a new connection: [Errno -3]
Temporary failure in name resolution')': /simple/scikit-learn/
ERROR: Could not find a version that satisfies the requirement argparse
(from unittest2) (from versions: none)
ERROR: No matching distribution found for argparse
./install_conda.sh: line 33: conda: command not found

Why it will error in this two command?
Does something I forgot to download or I enter the command under the wrong
path?

mguthaus, 2023-07-26:
You have a network problem and it cannot download things.

haotommy, 2023-07-27:
So I need to reinstall all?

mguthaus, 2023-07-27:
This is beyond the scope of OpenRAM. You haven't installed any of them because they didn't download.