VLSIDA/OpenRAM #203: Invalid column decoder encountered when generating large-scale SRAM

Chuld, 2023-08-06:
Thank you so much for taking a look at my question.

ERROR: file column_decoder.py: line 118: Invalid column decoder?
Traceback (most recent call last):
  File "/home/ICer/jl32n22/design/tools/OpenRAM/sram_compiler.py", line 73, in <module>
    s = sram()
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/sram.py", line 56, in __init__
    self.s.create_netlist()
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/modules/sram_1bank.py", line 199, in create_netlist
    self.add_modules()

  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/modules/sram_1bank.py", line 465, in add_modules
    self.bank = factory.create("bank", sram_config=self.sram_config, module_name="bank")
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/modules/bank.py", line 49, in __init__
    self.create_netlist()
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/modules/bank.py", line 61, in create_netlist
    self.create_instances()
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/modules/bank.py", line 165, in create_instances
    self.create_column_decoder()
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/modules/bank.py", line 535, in create_column_decoder
    self.column_decoder = factory.create(module_type="column_decoder",
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/modules/column_decoder.py", line 32, in __init__
    self.create_netlist()
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/modules/column_decoder.py", line 39, in create_netlist
    self.add_modules()
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/modules/column_decoder.py", line 118, in add_modules
    debug.error("Invalid column decoder?", -1)
  File "/home/ICer/jl32n22/design/tools/OpenRAM/compiler/debug.py", line 47, in error
    assert return_value == 0
AssertionError
make[1]: *** [Makefile:85: scn4m_subm_1w1r_4_small.ok] Error 1
make: *** [Makefile:95: scn4m_subm_1w1r_4_small] Error 2

Other relevant logs are as follows:
Technology: scn4m_subm
Total size: 307200 bits
Word size: 4
Words: 76800
Banks: 1
RW ports: 0
R-only ports: 1
W-only ports: 1

[openram.sram_config/recompute_sizes]: Recomputing with words per row: 4800
[openram.sram_config/recompute_sizes]: Rows: 16 Cols: 19200
[openram.sram_config/recompute_sizes]: Row addr size: 4 Col addr size: 12 Bank addr size: 16
[openram.sram_config/compute_sizes]: Set SRAM Words Per Row=4800
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_global_bitcell_array 16 x 19200
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_local_bitcell_array 16 x 16
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_capped_replica_bitcell_array 16 x 16
[openram.modules.capped_replica_bitcell_array/__init__]: Creating scn4m_subm_1w1r_4_small_capped_replica_bitcell_array 16 x 16 rbls: [1, 1] left_rbl: [0] right_rbl: []
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_replica_bitcell_array 16 x 16
[openram.modules.replica_bitcell_array/__init__]: Creating scn4m_subm_1w1r_4_small_replica_bitcell_array 16 x 16 rbls: [1, 1] left_rbl: [0] right_rbl: []
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_bitcell_array 16 x 16
[openram.modules.bitcell_array/__init__]: Creating scn4m_subm_1w1r_4_small_bitcell_array 16 x 16
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_replica_column 18 x 1
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_dummy_array 1 x 16
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_dummy_array_0 1 x 17
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_dummy_array_1 1 x 17
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_dummy_array_2 20 x 1
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_dummy_array_3 20 x 1
[openram.modules.wordline_buffer_array/__init__]: Creating scn4m_subm_1w1r_4_small_wordline_buffer_array
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_local_bitcell_array_0 16 x 16
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_capped_replica_bitcell_array_0 16 x 16
[openram.modules.capped_replica_bitcell_array/__init__]: Creating scn4m_subm_1w1r_4_small_capped_replica_bitcell_array_0 16 x 16 rbls: [1, 1] left_rbl: [] right_rbl: []
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_replica_bitcell_array_0 16 x 16
[openram.modules.replica_bitcell_array/__init__]: Creating scn4m_subm_1w1r_4_small_replica_bitcell_array_0 16 x 16 rbls: [1, 1] left_rbl: [] right_rbl: []
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_bitcell_array_0 16 x 16
[openram.modules.bitcell_array/__init__]: Creating scn4m_subm_1w1r_4_small_bitcell_array_0 16 x 16
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_dummy_array_4 1 x 16
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_dummy_array_5 1 x 16
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_dummy_array_6 1 x 16
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_dummy_array_7 20 x 1
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_local_bitcell_array_1 16 x 16
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_capped_replica_bitcell_array_1 16 x 16
[openram.modules.capped_replica_bitcell_array/__init__]: Creating scn4m_subm_1w1r_4_small_capped_replica_bitcell_array_1 16 x 16 rbls: [1, 1] left_rbl: [] right_rbl: [1]
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_replica_bitcell_array_1 16 x 16
[openram.modules.replica_bitcell_array/__init__]: Creating scn4m_subm_1w1r_4_small_replica_bitcell_array_1 16 x 16 rbls: [1, 1] left_rbl: [] right_rbl: [1]
[openram.modules.bitcell_base_array/__init__]: Creating scn4m_subm_1w1r_4_small_replica_column_0 18 x 1
[openram.modules.and2_dec/__init__]: Creating and2_dec and2_dec
[openram.modules.and3_dec/__init__]: Creating and3_dec and3_dec
[openram.modules.and4_dec/__init__]: Creating and4_dec and4_dec
[openram.modules.wordline_driver_array/__init__]: Creating scn4m_subm_1w1r_4_small_wordline_driver_array
[openram.modules.wordline_driver/__init__]: Creating wordline_driver wordline_driver
[openram.modules.pbuf/__init__]: creating pbuf with size of 1200
[openram.modules.precharge_array/__init__]: Creating scn4m_subm_1w1r_4_small_precharge_array
[openram.modules.column_mux_array/__init__]: Creating scn4m_subm_1w1r_4_small_column_mux_array
[openram.modules.write_driver_array/__init__]: Creating scn4m_subm_1w1r_4_small_write_driver_array
[openram.modules.precharge_array/__init__]: Creating scn4m_subm_1w1r_4_small_precharge_array_0
[openram.modules.sense_amp_array/__init__]: Creating scn4m_subm_1w1r_4_small_sense_amp_array
[openram.modules.column_mux_array/__init__]: Creating scn4m_subm_1w1r_4_small_column_mux_array_0

It seems that there is a problem with the number of columns of my SRAM, how should I configure the parameters to ensure that I get an SRAM with a suitable number of rows and columns? In addition, I would like to ask, what is the maximum number of words supported by the current version of OpenRAM? I often encounter this problem when generating large-scale SRAM.

mguthaus, 2023-08-07:
This is a very big memory with a very small word size. It probably is running intoa  corner case.   Do you have your config file and which version of openram you are using (commit id)?

We really focus on generating smaller blocks that can be used in larger memory designs. We have a virtual banking option which will generate Verilog that can be synthesized to use these multiple banks. 

We don't really have a "maximum" per say. There is a limit on the hierarchical decoder since we have at most a nand4 in the pre-decoders and the final decoder. However, a single bank will run into limits long before that due to long bitlines and bitline leakage/noise margins. It is difficult to predict in each technology, so we leave it to the user to do functional simulation. Ultimately, we'd like to automate it better, but there's a reason memory design is hard. :)

closed 2023-08-07 by mguthaus

Chuld, 2023-08-10:
word_size = 4
num_words = 307200  # 640*480 words(addresses)
write_size = 4

local_array_size = 16  # number of columns per array
words_per_row = 19200  # number of addresses per line
num_spare_rows = 0  
num_spare_cols = 0  

num_rw_ports = 0  
num_r_ports = 1  
num_w_ports = 1  

tech_name = "scn4m_subm"  
nominal_corner_only = True  

route_supplies = "minimal"
check_lvsdrc = True
perimeter_pins = False

output_name = "sram_{0}rw{1}r{2}w_{3}_{4}_{5}".format(num_rw_ports,
                                                      num_r_ports,
                                                      num_w_ports,
                                                      word_size,
                                                      num_words,
                                                      tech_name)
output_path = "macro/{}".format(output_name)

Here is my config file, there seems to be something wrong with the setting of local_array_size = 16 and words_per_row?
Also, while inspecting the device, I found out that the output of the write driver is made of two nmos that turn on gnd and bl/br at en din' and en din respectively, which doesn't seem to work if gnd remains grounded bl/br output high level? Will this cause failure to build SRAM?

mguthaus, 2023-08-11:
You are manually specifying a lot of parameters that aren't usually. I suggest not appreciating the words per row if you specify the 16 local arrays. I have never done that so it's likely the issue.