VLSIDA/OpenRAM #207: sram_compiler.py - python 3.8 posixpath error

htm23x, 2023-10-26:
Hi,
I installed openram in Mint 20.3 using pip3 (openram python files wwere deployed to my HOME folder in /home/myusername/.local/lib/python3.8/site-packages/openram ).

Then, I a simple config file... but I'm getting this error related to Pyhton 3.8.18 "posixpath.py":

`python3 sram_compiler.py -vv -n /var/tmp/openram_myconfig.py 
WARNING: file globals.py: line 168: Failed to find coverage installation. This can be installed with pip3 install coverage
[openram.globals/init_openram]: Initializing OpenRAM...
[openram.globals/setup_paths]: Setting up paths...
[openram.globals/setup_paths]: OpenRAM source code found in /home/myusername/.local/lib/python3.8/site-packages/openram/compiler
[openram.globals/setup_paths]: Temporary files saved in /tmp/openram_yo_68177_temp/
[openram.globals/read_config]: Configuration file is /var/tmp/openram_myconfig.py
[openram.globals/read_config]: Output saved in /home/myusername/.local/lib/python3.8/site-packages/openram/dist/
[openram.globals/install_conda]: Creating conda setup...
[openram.globals/import_tech]: Importing technology: sky130A
[openram.globals/import_tech]: Tech directory found in /home/myusername/.local/lib/python3.8/site-packages/openram/technology:/home/myusername/.local/lib/python3.8/site-packages/openram/technology
[openram.globals/import_tech]: Adding technology path: /home/myusername/.local/lib/python3.8/site-packages/openram/technology
[openram.globals/import_tech]: Adding technology path: /home/myusername/.local/lib/python3.8/site-packages/openram/technology
Traceback (most recent call last):
  File "sram_compiler.py", line 42, in <module>
    openram.init_openram(config_file=args[0])
  File "/home/myusername/.local/lib/python3.8/site-packages/openram/compiler/globals.py", line 190, in init_openram
    import_tech()
  File "/home/myusername/.local/lib/python3.8/site-packages/openram/compiler/globals.py", line 573, in import_tech
    OPTS.openram_tech = os.path.dirname(tech_mod.__file__) + "/"
  File "/home/myusername/.local/lib/python3.8/site-packages/openram/miniconda/lib/python3.8/posixpath.py", line 152, in dirname
    **p = os.fspath(p)**
**TypeError: expected str, bytes or os.PathLike object, not NoneType**

----
(base) myusername@COMPUTER:~/.local/lib/python3.8/site-packages/openram$ python3 --version
Python 3.8.18

----
(base) myusername@COMPUTER:~/.local/lib/python3.8/site-packages/openram$ lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 20.3
Release:	20.3
Codename:	una`

mguthaus, 2023-10-26:
Hi, 
Can you try using Python 3.8.10? I don't have an installation of Mint linux and it isn't really supported.

mguthaus, 2023-10-26:
What is in your config file?

htm23x, 2023-10-26:
Bingo! I was using this param in the config file:

`tech_name="sky130A"`

which is incorrect ( should be just "sky130", without the final "A" letter ), so python complained.
You should add some validation to the config params to avoid things like that (as the python error message is a bit confusing ).

htm23x, 2023-10-26:
Closing issue and creating a pull request #210 to improve tech_name param validation before the module is imported. Thanks.

closed 2023-10-26 by htm23x