VLSIDA/OpenRAM #208: Support for IHP PDK

htm23x, 2023-10-26:
Hi, I was wondering if could be possible to add IHP-Open-PDK support, please. Thanks.

[](https://github.com/IHP-GmbH/IHP-Open-PDK)

mguthaus, 2023-10-26:
We would like to add support, but it requires significant work. We would be glad to take this as a contribution.

mguthaus, 2026-10-13:
Status update: there is a work-in-progress port in #302 (currently a draft PR) for ihp13sg2.

Leaving this open to track it. If you would like to help, reviewing or building on that PR is the most useful thing right now — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md) and [technology_setup.md](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/technology_setup.md) for what a technology port involves.

bhagwat-rahul, 2026-10-26:
#302 is ready for review