VLSIDA/OpenRAM #209: 4 test case failed for freepdk and scn4m

z4fang, 2023-10-26:
**Describe the bug**
A clear and concise description of what the bug is.

FAILING TESTS
freepdk45/21_model_delay_test.bad
freepdk45/21_xyce_setuphold_test.bad
freepdk45/23_lib_sram_test.bad
freepdk45/30_openram_sram_char_test.bad
scn4m_subm/21_model_delay_test.bad
scn4m_subm/21_xyce_setuphold_test.bad
scn4m_subm/23_lib_sram_test.bad
scn4m_subm/30_openram_sram_char_test.bad

all logs have this error: 

`[2023-10-26 10:13:54.741365] ERROR: file stimuli.py: line 421: Spice simulation error: /bin/bash -c 'source /home/kfang/work/unittest/OpenRAM/miniconda/bin/activate && /home/kfang/work/unittest/OpenRAM/miniconda/bin/mpirun -np 8 /home/kfang/work/unittest/OpenRAM/miniconda/bin/Xyce -r /home/kfang/work/unittest/OpenRAM/compiler/tests/results/scn4m_subm/21_model_delay_test/tmp/timing.raw -o /home/kfang/work/unittest/OpenRAM/compiler/tests/results/scn4m_subm/21_model_delay_test/tmp/timing.lis /home/kfang/work/unittest/OpenRAM/compiler/tests/results/scn4m_subm/21_model_delay_test/tmp/delay_stim.sp && conda deactivate'
`

**Version**
Which commit are you using?
3113798b13dac5470534fde5aaebbb038dce195f

**To Reproduce**
What did you do to demonstrate the bug?
Please include your configuration file used.

**Logs**
If applicable, add logs or output to help explain your problem.

[image]

**Additional context**
Add any other context about the problem here.

mguthaus, 2023-10-26:
It would help to see the log files for each of the tests to see what is going on. These should be in the directory with .log extension.

Since these are all characterization, it is likely an updated version of Xyce that causes a characterization problem.

z4fang, 2023-10-26:
[logs.zip](https://github.com/VLSIDA/OpenRAM/files/13180404/logs.zip)

I have attached the .log and .out logs.
I belive the error come from the command 
`/home/kfang/work/OpenRAM/env/bin/mpirun -np 3 /home/kfang/work/OpenRAM/env/bin/Xyce -r /tmp/openram_kfang_2312715_temp/timing.raw -o /tmp/openram_kfang_2312715_temp/timing.lis /tmp/openram_kfang_2312715_temp/delay_stim.sp`

[image]

mguthaus, 2023-10-26:
It looks like your system is missing a Xyce dependency. You need to install the fftw3-3 shared library. In Ubuntu, I just do:
`sudo apt install fftw3`

erendn, 2023-12-20:
This issue should be fixed with v1.2.43 (https://github.com/VLSIDA/OpenRAM/commit/caa9b2a2d87d2c8f088c9e8807d5fe0aab70a4c5).

closed 2023-12-20 by erendn