VLSIDA/OpenRAM #211: Pin iterator not found error in hierarchy_layout.py

htm23x, 2023-10-26:
Hi,
I'm trying to generate a 1Kbyte (1024 words x 8bit x 1bank x 1port) SRAM using sky130, but openram complains about it with a pin iterator/pin not found error. Here's the log and my config file

----
```
(base) myUser@COMPUTER:~/.local/lib/python3.8/site-packages/openram$ python3 sram_compiler.py /var/tmp/myconfig.py 
|==============================================================================|
|=========                      OpenRAM v1.2.40                       =========|
|=========                                                            =========|
|=========               VLSI Design and Automation Lab               =========|
|=========        Computer Science and Engineering Department         =========|
|=========            University of California Santa Cruz             =========|
|=========                                                            =========|
|=========          Usage help: openram-user-group@ucsc.edu           =========|
|=========        Development help: openram-dev-group@ucsc.edu        =========|
|=========                See LICENSE for license info                =========|
|==============================================================================|
** Start: 10/27/2023 00:13:18
Technology: sky130
Total size: 8192 bits
Word size: 8
Words: 1024
Banks: 1
RW ports: 1
R-only ports: 0
W-only ports: 0
DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).
Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
Words per row: None
Output files are: 
/home/myUser/.local/lib/python3.8/site-packages/openram/dist/my_sram_1kb_1024word_8bit_1bank_1port.lvs
/home/myUser/.local/lib/python3.8/site-packages/openram/dist/my_sram_1kb_1024word_8bit_1bank_1port.sp
/home/myUser/.local/lib/python3.8/site-packages/openram/dist/my_sram_1kb_1024word_8bit_1bank_1port.v
/home/myUser/.local/lib/python3.8/site-packages/openram/dist/my_sram_1kb_1024word_8bit_1bank_1port.lib
/home/myUser/.local/lib/python3.8/site-packages/openram/dist/my_sram_1kb_1024word_8bit_1bank_1port.py
/home/myUser/.local/lib/python3.8/site-packages/openram/dist/my_sram_1kb_1024word_8bit_1bank_1port.html
/home/myUser/.local/lib/python3.8/site-packages/openram/dist/my_sram_1kb_1024word_8bit_1bank_1port.log
/home/myUser/.local/lib/python3.8/site-packages/openram/dist/my_sram_1kb_1024word_8bit_1bank_1port.lef
/home/myUser/.local/lib/python3.8/site-packages/openram/dist/my_sram_1kb_1024word_8bit_1bank_1port.gds
ERROR: file hierarchy_layout.py: line 579: Should use a pin iterator since more than one pin vpb
ERROR: file hierarchy_layout.py: line 586: No pin found with name vpb on sky130_fd_bd_sram__sram_sp_colenda. Saved as missing_pin.gds.
Traceback (most recent call last):
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/base/hierarchy_layout.py", line 579, in get_pin
    debug.error("Should use a pin iterator since more than one pin {}".format(text), -1)
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/debug.py", line 48, in error
    assert return_value == 0
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "sram_compiler.py", line 73, in <module>
    s = sram()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/sram.py", line 56, in __init__
    self.s.create_netlist()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/modules/sram_1bank.py", line 198, in create_netlist
    self.add_modules()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/modules/sram_1bank.py", line 457, in add_modules
    self.bank = factory.create("bank", sram_config=self.sram_config, module_name="bank")
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/modules/bank.py", line 49, in __init__
    self.create_netlist()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/modules/bank.py", line 59, in create_netlist
    self.add_modules()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/modules/bank.py", line 392, in add_modules
    self.bitcell_array = factory.create(module_type="capped_replica_bitcell_array",
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/technology/sky130/custom/sky130_capped_replica_bitcell_array.py", line 47, in __init__
    self.create_netlist()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/technology/sky130/custom/sky130_capped_replica_bitcell_array.py", line 53, in create_netlist
    self.add_modules()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/technology/sky130/custom/sky130_capped_replica_bitcell_array.py", line 58, in add_modules
    self.replica_bitcell_array = factory.create(module_type="replica_bitcell_array",
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/technology/sky130/custom/sky130_replica_bitcell_array.py", line 61, in __init__
    super().__init__(self.row_size, self.column_size, rbl, left_rbl, right_rbl, name)
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/technology/sky130/custom/replica_bitcell_array.py", line 72, in __init__
    self.create_netlist()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/technology/sky130/custom/replica_bitcell_array.py", line 82, in create_netlist
    self.add_modules()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/technology/sky130/custom/replica_bitcell_array.py", line 136, in add_modules
    self.replica_columns[port] = factory.create(module_type="replica_column",
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/sram_factory.py", line 153, in create
    obj = mod(name=module_name, **kwargs)
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/technology/sky130/custom/sky130_replica_column.py", line 63, in __init__
    self.create_layout()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/technology/sky130/custom/sky130_replica_column.py", line 76, in create_layout
    self.add_layout_pins()
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/technology/sky130/custom/sky130_replica_column.py", line 240, in add_layout_pins
    pin = inst.get_pin("vpb")
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/base/geometry.py", line 265, in get_pin
    pin = copy.deepcopy(self.mod.get_pin(name))
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/base/hierarchy_layout.py", line 586, in get_pin
    debug.error("No pin found with name {0} on {1}. Saved as missing_pin.gds.".format(name, self.cell_name), -1)
  File "/home/myUser/.local/lib/python3.8/site-packages/openram/compiler/debug.py", line 48, in error
    assert return_value == 0
AssertionError
```

----
/var/tmp/myconfig.py 

```
num_threads = 4

num_words = 1024
word_size = 8
num_rw_ports = 1
num_banks = 1

tech_name = "sky130"
num_spare_rows = 1
num_spare_cols = 1

process_corners = [ "TT" ]
supply_voltages = [ 3.3 ]
temperatures = [ 25 ]

output_path = "dist"
output_name = "my_sram_1kb_1024word_8bit_1bank_1port"
```

htm23x, 2023-10-28:
OpenRAM v1.2.40 using "pip3 install openram" (yesterday).
I'll check later the latest git master later to see if the error was solved.

htm23x, 2023-10-28:
I've tried with pip3, 1.2.40 branch, master branch & dev branch with same result. I think there's a bug and it's impossible to compile an SRAM with sky130 tech using 1 port without getting tha pin error message. You should revise the code.

mguthaus, 2023-10-29:
Is this the same issue as #204?

Do you have the most recent sky130_sram_fd_bd library cloned?

htm23x, 2023-10-29:
How I do that, pls? I just cloned the repo yesterday and followed the build instructions from the documentation. Not sure how to do that.

htm23x, 2023-10-29:
I tried to update the sky130_sram_fd_bd lib using a "git pull origin main" and fetched this commit
 [flatten colends](https://github.com/VLSIDA/sky130_fd_bd_sram/commit/118c1a628c5f8ffb58a8e1a20de2b67c5058345f)
from user [jcirimel](https://github.com/VLSIDA/sky130_fd_bd_sram/commits?author=jcirimel) did 2h ago... but got the same pin error message.

jcirimel, 2023-10-29:
Sorry I have this fixed upstream, but with a bunch of other changes that don't think I'm going to be able to merge for a bit. Let me see if I can make a small fix for the current dev branch for this.

jcirimel, 2023-10-29:
Sorry I closed this by accident

htm23x, 2023-10-29:
Don't worry, take your time, there's no hutty. I'll chack later. Many thanks.

jcirimel, 2023-11-01:
Ok I put in a fix for this. We did recently find a bug in our sky130 single port macros affecting all past and current sky130 single port designs. So I would not use a single port macro right now, even if it is passing lvs (although it probably won't be passing now).

jcirimel, 2023-11-01:
It might take a few hours for the fix to show up in this repo

closed 2023-11-01 by jcirimel