VLSIDA/OpenRAM #212: scn3me_subm tech CellProperties error

htm23x, 2023-10-29:
I'm trying to compile a 1Kb x 8 bit SRAM using the scn3me_subm technology, but I got an "import CellProperties" error (works ok with the scn4m_subm technology though! ). Here's the log and config file:

----
```
(base) myUser@COMPUTER:/var/tmp/OpenRAM-1.2.40$ python3 sram_compiler.py /var/tmp/myconfig.py

ERROR: file globals.py: line 581: Could not load tech module.
Traceback (most recent call last):
  File "/var/tmp/OpenRAM-1.2.40/compiler/globals.py", line 579, in import_tech
    from openram import tech
  File "/var/tmp/OpenRAM-1.2.40/technology/scn3me_subm/tech/__init__.py", line 5, in <module>
    from .tech import *
  File "/var/tmp/OpenRAM-1.2.40/technology/scn3me_subm/tech/tech.py", line 4, in <module>
    from openram.drc.custom_cell_properties import CellProperties
ImportError: cannot import name 'CellProperties' from 'openram.drc.custom_cell_properties' (/var/tmp/OpenRAM-1.2.40/compiler/drc/custom_cell_properties.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "sram_compiler.py", line 42, in <module>
    openram.init_openram(config_file=args[0])
  File "/var/tmp/OpenRAM-1.2.40/compiler/globals.py", line 190, in init_openram
    import_tech()
  File "/var/tmp/OpenRAM-1.2.40/compiler/globals.py", line 581, in import_tech
    debug.error("Could not load tech module.", -1)
  File "/var/tmp/OpenRAM-1.2.40/compiler/debug.py", line 48, in error
    assert return_value == 0
```

----
Config file:

```
num_words = 1024
word_size = 8

num_rw_ports = 1
num_banks = 1

tech_name = "scn3me_subm"

num_spare_rows = 0
num_spare_cols = 0

process_corners = [ "TT" ]
supply_voltages = [ 3.3 ]
temperatures = [ 25 ]

output_path = "dist"
output_name = "my_sram_1kb_1024word_8bit_1bank_1port"

```

mguthaus, 2023-10-29:
This technology is no longer functional.

mguthaus, 2023-10-29:
(it only has three metal layers...)

closed 2023-10-29 by mguthaus

htm23x, 2023-10-29:
Oh I see! Why don't remove the tech folder "scn3me_subm" ? Save space in repo and avoid confusion :D

mguthaus, 2023-10-29:
Someone may want the cells...