VLSIDA/OpenRAM #213: Generating bit cell array with custom bit cell.

tomaschoi03, 2023-11-02:
I'm trying to generate an sram with custom bit cells. The bit cells are 6T with two word lines, one connected to the left transistor and another connected to the right transistor. The way I implemented this custom bit cell was by taking the `cell_1rw.gds` file in the `technology/freepdk45` directory and modified it on Virtuoso using the layout tools. Is this the correct way to go about creating a custom bit cell? After saving this modified bit cell gds file, I changed the configuration file by adding `bitcell = "cell_1rw_2wl"`  thinking that it would just take that gds file and create an array. After running the compiler, the output sram still uses the `cell_1rw.gds` file in the `technology/freepdk45`. Is there a step I'm missing?

mguthaus, 2023-11-02:
That sounds like the correct step. Are you positive it is using your library and not another one (for example, with the pip installation).

How did you install OpenRAM? Did you set the tech variable?

Some of the paths should be shown in the log output. You may have to run with "-v" to get additional information.

tomaschoi03, 2023-11-06:
I cloned the repository and followed the steps in the basic_usage.md page. I set `export OPENRAM_TECH="$HOME/openram/technology" `. 

When I run the sram compiler, it does show that it is pointing to the technology/freepdk45 library. 

In the basic_usage.md page it says that to use custom bit cells, we need to add a `bitcell` and a `replica_bitcell` variable. What is the difference between the two? Or is it really just an exact replica? Also, I just want to confirm that with the custom bitcell we only need the gds file? Or do we need other types of files in the technology directory like spice files?

mguthaus, 2023-11-07:
1) You need a bitcell, dummy bitcell and replica bitcell. The dummy should have its access transistors disabled so it just loads the word line. The replica cell should always drive a 0 on the bitline (and 1 on bitline bar). These are used for the replica array timing.

2) you also need spice models for each for both LVS and simulation. These might be the same.

mguthaus, 2023-11-07:
You also need a sense amp and some other cells like possibly decodes depending on the architecture.

closed 2023-11-24 by tomaschoi03