VLSIDA/OpenRAM #214: Virtuoso Layout using freepdk45

tomaschoi03, 2023-11-24:
I'm trying to modify a layout of one of the standard cells in the freepdk45. I'm having trouble adding a new library into Cadence Virtuoso using the technology file found in `OpenRAM/technology/freepdk45/tf/FreePDK45.tf`. The following is a piece of the warning/error messages I see. Does anyone have a similar problem? And how did they fix it?

```
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal4 track).
*WARNING* (TECH-280074): Layer metal5 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal5 track).
*WARNING* (TECH-280074): Layer metal6 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal6 track).
*WARNING* (TECH-280074): Layer metal7 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal7 track).
*WARNING* (TECH-280074): Layer metal8 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal8 track).
*WARNING* (TECH-280074): Layer metal9 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal9 track).
*WARNING* (TECH-280074): Layer metal10 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal10 track).
*WARNING* (TECH-280074): Layer contact was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (contact grid).
*WARNING* (TECH-280074): Layer metal1 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal1 grid).
*WARNING* (TECH-280074): Layer via1 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (via1 grid).
*WARNING* (TECH-280074): Layer metal2 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal2 grid).
*WARNING* (TECH-280074): Layer via2 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (via2 grid).
*WARNING* (TECH-280074): Layer metal3 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal3 grid).
*WARNING* (TECH-280074): Layer via3 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (via3 grid).
*WARNING* (TECH-280074): Layer metal4 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal4 grid).
*WARNING* (TECH-280074): Layer via4 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (via4 grid).
*WARNING* (TECH-280074): Layer metal5 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal5 grid).
*WARNING* (TECH-280074): Layer via5 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (via5 grid).
*WARNING* (TECH-280074): Layer metal6 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal6 grid).
*WARNING* (TECH-280074): Layer via6 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (via6 grid).
*WARNING* (TECH-280074): Layer metal7 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal7 grid).
*WARNING* (TECH-280074): Layer via7 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (via7 grid).
*WARNING* (TECH-280074): Layer metal8 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal8 grid).
*WARNING* (TECH-280074): Layer via8 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (via8 grid).
*WARNING* (TECH-280074): Layer metal9 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal9 grid).
*WARNING* (TECH-280074): Layer via9 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (via9 grid).
*WARNING* (TECH-280074): Layer metal10 was not defined.
*WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal10 grid).
*WARNING* (TECH-280074): Layer pwell was not defined.
*WARNING* (TECH-280085): *ERROR* Failed to define layer display for (pwell drawing pwell t t t t t).
*WARNING* (TECH-280074): Layer nwell was not defined.
*WARNING* (TECH-280085): *ERROR* Failed to define layer display for (nwell drawing nwell t t t t t).
*WARNING* (TECH-280074): Layer vtg was not defined.
*WARNING* (TECH-280085): *ERROR* Failed to define layer display for (vtg drawing vtg t t t t t).
*WARNING* (TECH-280074): Layer vth was not defined.
*WARNING* (TECH-280085): *ERROR* Failed to define layer display for (vth drawing vth t t t t t).
*WARNING* (TECH-280074): Layer active was not defined.
*WARNING* (TECH-280085): *ERROR* Failed to define layer display for (active drawing active t t t t t).
*WARNING* (TECH-280074): Layer nimplant was not defined.
*WARNING* (TECH-280085): *ERROR* Failed to define layer display for (nimplant drawing nimplant t t t t t).
*WARNING* (TECH-280074): Layer pimplant was not defined.
*WARNING* (TECH-280085): *ERROR* Failed to define layer display for (pimplant drawing pimplant t t t t t).
[... 395 lines cut ...]
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal6).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal6).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal6).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal7).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal7).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal7).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal7).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal7).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal7).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal7).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal7).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal8).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal8).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal8).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal8).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal8).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal8).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal9).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal9).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal9).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal9).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal10).
*WARNING* (TECH-2000192): techLayerProperties: Invalid layer/purpose - (metal10).
*WARNING* (TECH-280089): techDerivedLayers: Cannot create the derived layer 'noOverlapLayer1' based on layer 'poly' - 'poly' is not defined.
```

mguthaus, 2023-11-24:
I would suggest the freepdk mailing list which you can find here:
https://eda.ncsu.edu/freepdk/freepdk45/

We just include that for convenience.

closed 2023-11-24 by mguthaus

stineje, 2023-11-24:
Hi,

Is it is possible, you didn’t set up your cds.lib correctly? Also, check to make sure you make your library correctly (i.e., attached your tech file).  You should also be able to create a sample layout in your Library Manager and you should see all the layers in Virtuoso.  If you do not see any layers when you open up a blank layout, you did not set the library up correctly.  

All my best,

James

> On Nov 24, 2023, at 2:15 PM, tomaschoi03 ***@***.***> wrote:
> 
> Failed to define layer-purpose priority for (metal4 track).
> *WARNING* (TECH-280074): Layer metal5 was not defined.
> *WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal5 track).
> *WARNING* (TECH-280074): Layer metal6 was not defined.
> *WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal6 track).
> *WARNING* (TECH-280074): Layer metal7 was not defined.
> *WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal7 track).
> *WARNING* (TECH-280074): Layer metal8 was not defined.
> *WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal8 track).
> *WARNING* (TECH-280074): Layer metal9 was not defined.
> *WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal9 track).
> *WARNING* (TECH-280074): Layer metal10 was not defined.
> *WARNING* (TECH-280082): *ERROR* Failed to define layer-purpose priority for (metal10 track).
> *WARNING* (TECH-280074): Layer contact was not defined.
> *WARNING* (TECH-280082): *ERROR* Failed to define lay