VLSIDA/OpenRAM #216: ROM signal names do not match model

mole99, 2023-12-16:
**Describe the bug**

Hello! It seems that the naming of the signals in the generated ROM models does not match those in the actual layout/LEF file.

| model | layout |
|-------|--------|
| clk0  | clk    |
| csb0  | cs     |
| addr0 | addr   |
| dout0 | dout   |
| vccd1 | vdd    |
| vssd1 | gnd    |

I believe the Verilog model was adjusted from the SRAM model early during development, but was not adapted to the actual layout.

The question is what naming should be used? Will there be two port ROMs in the near future? Then it would probably make sense to keep the `0` at the end of the signals. But then `csb0` must be named `cs0`.

If the names from the layout are to be used, then I think I can easily adapt the Verilog model generation to use the names from the layout.

If I adapt the Verilog model to match the layout and use #215, I am able to integrate the ROM using OpenLane.

**Version**
Latest stable/dev.

**To Reproduce**
Generate a ROM.

**Expected behavior**
Signal names match.

**Logs**

**Additional context**

hadirkhan10, 2023-12-21:
@mole99 the issue should now be fixed with the latest commit

mole99, 2023-12-21:
Yes, the signal names now match. Thank you!

Together with #215, integration with OpenLane works without any changes to the model or layout.

closed 2023-12-21 by hadirkhan10