VLSIDA/OpenRAM #217: LVS mismatch errors in SKY130 SRAM

KEVIN09876, 2023-12-18:
I am encountering LVS mismatch issues while generating a single-port SRAM using the Sky130 technology.
Could the problem be related to num_spare_col and num_spare_row in our config? How can I address and resolve this issue?

Another issue I'm facing is that the spare_wen pin is not in the lib file.
Also, the power values in the lib file appear to be excessively large, around 10^9.
Can you provide guidance on how to address this?

My OpenRAM version is v1.2.45.
 |==============================================================================|
 |=========                      OpenRAM v1.2.42                       =========|
 |=========                                                            =========|
 |=========               VLSI Design and Automation Lab               =========|
 |=========        Computer Science and Engineering Department         =========|
 |=========            University of California Santa Cruz             =========|
 |=========                                                            =========|
 |=========          Usage help: openram-user-group@ucsc.edu           =========|
 |=========        Development help: openram-dev-group@ucsc.edu        =========|
 |=========                See LICENSE for license info                =========|
 |==============================================================================|
 ** Start: 12/18/2023 02:25:16
 Technology: sky130
 Total size: 128 bits
 Word size: 8
Words: 16
Banks: 1
RW ports: 1
R-only ports: 0
W-only ports: 0
 DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
 Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
 Only generating nominal corner timing.
 Words per row: None
 Output files are:
 /home/user/OpenRAM/result/sram_1rw_8_16_sky130/sram_1rw_8_16_sky130.lvs
 /home/user/OpenRAM/result/sram_1rw_8_16_sky130/sram_1rw_8_16_sky130.sp
 /home/user/OpenRAM/result/sram_1rw_8_16_sky130/sram_1rw_8_16_sky130.v
 /home/user/OpenRAM/result/sram_1rw_8_16_sky130/sram_1rw_8_16_sky130.lib
 /home/user/OpenRAM/result/sram_1rw_8_16_sky130/sram_1rw_8_16_sky130.py
 /home/user/OpenRAM/result/sram_1rw_8_16_sky130/sram_1rw_8_16_sky130.html
 /home/user/OpenRAM/result/sram_1rw_8_16_sky130/sram_1rw_8_16_sky130.log
 /home/user/OpenRAM/result/sram_1rw_8_16_sky130/sram_1rw_8_16_sky130.lef
 /home/user/OpenRAM/result/sram_1rw_8_16_sky130/sram_1rw_8_16_sky130.gds
 ** Submodules: 2.9 seconds
 ** Placement: 0.0 seconds
 ** Routing: 158.3 seconds
 ERROR: file magic.py: line 385: sram_1rw_8_16_sky130       LVS mismatch (results in /tmp/openram_root_24133_temp/sram_1rw_8_16_sky130.lvs.report)

 ** Verification: 169.8 seconds
 ** SRAM creation: 331.1 seconds
 SP: Writing to /home/user/OpenRAM/result/sram_1rw_8_16_sky130/sram_1rw_8_16_sky130.sp
 ** Spice writing: 0.1 seconds

d-m-bailey, 2024-02-28:
@KEVIN09876 Can you share `sram_1rw_8_16_sky130.lvs.report `?

mguthaus, 2026-10-13:
Closing as stale. This has been inactive for 2.6 years and there is no current reproduction against `dev`, so it is not actionable as it stands.

Superseded in practice by #220 and #292, which cover sky130 LVS mismatches with current detail and are still open.

To be clear, this is being closed for inactivity rather than rejected. If it still affects you, please reopen or file a fresh issue with a reproduction against current `dev`, including your configuration file, technology, OS and `python3 --version`, and the contents of the temp directory OpenRAM leaves behind on failure ([the FAQ explains which files those are](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself, a PR would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch.

closed 2026-10-13 by mguthaus