VLSIDA/OpenRAM #218: Use Volare as PDK version manager

mole99, 2023-12-23:
Volare is a version manager and builder for PDKs using open_pdks. https://github.com/efabless/volare

I believe it would be beneficial for the OpenRAM project to use Volare instead of the current Makefile + open_pdks approach. 
One improvement is that PDK versions that have been built upstream only need to be enabled (downloaded), if not, the PDK will be built using open_pdks. This saves time and disk space. It is also much easier to change the PDK version. This is useful for when OpenRAM needs to be updated to newer PDK versions.

Volare can be installed via pip together with the other Python modules. Thanks to its Python API, Volare can be integrated directly into OpenRAM.

Currently sky130 and gf180mcu supported.

Projects that use volare:

- OpenLane 1: https://github.com/The-OpenROAD-Project/OpenLane
- OpenLane 2: https://github.com/efabless/openlane2
- DFFRAM: https://github.com/AUCOHL/DFFRAM

As a first step, I would look into enabling volare in the Makefile.

mguthaus, 2023-12-23:
We actually use conda right now.

mole99, 2023-12-23:
But not for PDK management :)

Currently OpenRAM installs all tools via conda and then goes on to build the PDK using open_pdks.

Volare can also be installed using conda + pip and then be used for PDK management. You can see this approach in PR #219.

It integrates very well with your current infrastructure.

mguthaus, 2023-12-23:
Yes. I'll consider this after the break. This is the first day of vacation right now. :)

mole99, 2023-12-23:
I did not expect anything different so close to Christmas ^^
Thanks, have a nice vacation!

erendn, 2024-01-19:
Done #219.

closed 2024-01-19 by erendn