VLSIDA/OpenRAM #220: LVS mismatch for some sky130 example SRAMs

ubfx, 2023-12-27:
**Describe the bug**
When compiling some ("sky130_sram_1rw_tiny", "sky130_sram_1kbyte_1rw_32x256_8", maybe others too) sky130 example srams from the "macros/sram_configs" subfolder, an LVS mismatch occurs.

**Version**
tag v1.2.45, commit fc37bf685991b9c9281b6d2fba6376ab0afc5231

**To Reproduce**
1. Install tools through conda and PDKs using the provided scripts ("install_conda.sh", "make pdk", "make install")
2. `cd macros && make sky130_sram_1rw_tiny.ok`

**Expected behavior**
Expect the LVS to conclude without mismatches, just like it works with other example configurations.

**Logs**
```
...
[2023-12-27 17:19:46.675595] [openram.verify.magic/run_lvs]: Cell pin lists for sky130_sram_1rw_tiny and sky130_sram_1rw_tiny altered to match.
[2023-12-27 17:19:46.676333] [openram.verify.magic/run_lvs]: Device classes sky130_sram_1rw_tiny and sky130_sram_1rw_tiny are equivalent.
[2023-12-27 17:19:46.677174] [openram.verify.magic/run_lvs]: 
[2023-12-27 17:19:46.677948] [openram.verify.magic/run_lvs]: Final result: Top level cell failed pin matching.
[2023-12-27 17:19:46.678774] ERROR: file magic.py: line 385: sky130_sram_1rw_tiny	LVS mismatch (results in /user/projekt_abc/OpenRAM/macros/sky130_sram_1rw_tiny/tmp/sky130_sram_1rw_tiny.lvs.report)
```

[Complete log: sky130_sram_1rw_tiny.log](https://github.com/VLSIDA/OpenRAM/files/13779912/sky130_sram_1rw_tiny.log)

ShenShan123, 2025-03-11:
I have the same problem in the latest stable version. When using the sky130 technology to generate sky130_sram_1rw_tiny, I got LVS mismatch reported as `Circuit 1 contains 1792 devices, Circuit 2 contains 1783 devices. *** MISMATCH ***
Circuit 1 contains 658 nets,    Circuit 2 contains 658 nets.` in lvs.out and ` NET mismatches: Class fragments follow (with fanout counts):
Circuit 1: sky130_sram_1rw_tiny_sky130_col_cap_array_0                                                                    |Circuit 2: sky130_sram_1rw_tiny_sky130_col_cap_array_0
DEVICE mismatches: Class fragments follow (with node fanout counts):
Circuit 1: sky130_sram_1rw_tiny_sky130_col_cap_array                                                                      |Circuit 2: sky130_sram_1rw_tiny_sky130_col_cap_array 
DEVICE mismatches: Class fragments follow (with node fanout counts):
Circuit 1: sky130_sram_1rw_tiny_sky130_replica_column                                                                     |Circuit 2: sky130_sram_1rw_tiny_sky130_replica_column
... `
How can I fix this bug?

mguthaus, 2025-07-14:
@jcirimel is yours passing?

mguthaus, 2026-10-13:
Still open. #292 covers the same ground with more recent detail and is also open, so it is worth reading alongside this.

If you are still chasing it: LVS runs through `compiler/verify/magic.py` (Magic extraction plus netgen), and the extracted netlist and the netgen report are both left in the temp directory on failure. Note that #320 just changed the extraction step on `dev` — it no longer extracts parasitics for the LVS netlist, since `ext2spice` discarded them anyway — so the `.ext` files will look different, though the netgen verdict should not change.

If you would like to take it, a PR against `dev` would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch, and the regression suite is run with `cd compiler/tests && make -k -j<N>`.