VLSIDA/OpenRAM #221: How to generate an SRAM with more than two ports?

HotCoCoC, 2024-01-10:
Hi, this is a great job for templates that want to generate custom RAM. I'm very grateful to see this project because I think it will reduce my verification time a lot. Unfortunately, the current OpenRAM only supports up to 2 ports of RAM. If I want to generate a 4-port RAM such as 1rw, 3r how do I change this code? I notice that add_modules in bank.py seems to limit the number of ports.
[image]
But there's something I don't understand. What does the 1 in rbl stand for? What does the number of 1 in the list represent? Why is left_rbl 0 and right_rbl 1?

mguthaus, 2024-01-10:
More than two ports would venture into "register file" territory which often has different types of bitcells (with more ports). We only have bitcells with one and two ports.

The RBL is the replica bitline which has an extra bit that is hard wired to a value and used for timing the control logic. Left and right would be for two ports.

Hope that helps.

closed 2024-01-10 by mguthaus

manwu1994, 2024-02-13:
Do you have any ideas or solutions for generating an SRAM with more than two ports?

mguthaus, 2024-02-13:
You could use flip flops or latches instead.

manwu1994, 2024-02-15:
> You could use flip flops or latches instead.
Thank you so much for your reply. I read the paper "Automated Synthesis of Multi-Port Memories and Control",  in this end, can we directly use the parameter to define the multi-port, like 
```
num_rw_ports = 4
num_r_ports = 4
num_w_ports = 4
```
Furthermore, I have reviewed the option.py file, which contains the comment "# Port configuration (1-2 ports allowed)". Therefore, if I would like to generate a multi-port memory, can I directly use configuration parameters to define or build the custom multi-port "add_modules"?

Thank you in advance for your assistance and response.

mguthaus, 2024-02-15:
The intent is to make a register file compiler at some point but it isn't supported right now. The total ports must be less than 3 if you want a physical implementation. It can generate spice netlists with more than that but it isn't well tested since it isn't used.