VLSIDA/OpenRAM #226: Unable to read configuration file

cannotgetaname, 2024-02-04:
**Describe the bug**
executing like this:
python3 $OPENRAM_HOME/../sram_compiler.py myconfig

but errors like this:
ERROR: file globals.py: line 326: Unable to read configuration file: /home/zcz/graduationproject/OpenRAM/works/01/myconfig

I cloned the latest ones.

I used the exemple config files.

What I want to do is run the whole software to compile a sram.

all error info:
ERROR: file globals.py: line 326: Unable to read configuration file: /home/zcz/graduationproject/OpenRAM/works/01/myconfig
Traceback (most recent call last):
  File "/home/zcz/graduationproject/OpenRAM/compiler/globals.py", line 324, in read_config
    config = importlib.import_module(module_name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'myconfig'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zcz/graduationproject/OpenRAM/compiler/../sram_compiler.py", line 42, in <module>
    openram.init_openram(config_file=args[0])
  File "/home/zcz/graduationproject/OpenRAM/compiler/globals.py", line 186, in init_openram
    read_config(config_file, is_unit_test)
  File "/home/zcz/graduationproject/OpenRAM/compiler/globals.py", line 326, in read_config
    debug.error("Unable to read configuration file: {0}".format(config_file), 2)
  File "/home/zcz/graduationproject/OpenRAM/compiler/debug.py", line 48, in error
    assert return_value == 0
AssertionError

mguthaus, 2024-02-04:
It should have a .py file extension.

cannotgetaname, 2024-02-04:
thank you so much! It works!
I'm afraid that I left thr manual behind,sorry to bother you.
Please believe me,i really have RTFM.

mguthaus, 2024-02-04:
No problem!

closed 2024-02-04 by mguthaus