VLSIDA/OpenRAM #227: openram.py doesn't exist in any tree

msarwarx1, 2024-02-14:
**Describe the bug**
A clear and concise description of what the bug is.

**Version**
Which commit are you using?

**To Reproduce**
What did you do to demonstrate the bug?
Please include your configuration file used.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
If applicable, add logs or output to help explain your problem.

**Additional context**
Add any other context about the problem here.

mguthaus, 2024-02-14:
This was replaced in a later version to allow the separation of generation and characterization as well as different executables for different memories. See:

sram_compiler.py
rom_compiler.py
sram_char.py
sram_func.py

Where are you still seeing this? If there is outdated documentation, we should update it.

mguthaus, 2026-10-13:
Closing as obsolete.

`openram.py` no longer exists anywhere in the tree, and nothing in the docs or README refers to it any more. The current entry points at the repository root are:

- `sram_compiler.py` — SRAM generation
- `rom_compiler.py` — ROM generation
- `sram_char.py` — characterization only
- `sram_func.py` — functional simulation only

Usage is documented in [basic_usage.md](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/basic_usage.md).

If you are still hitting a stale reference somewhere, please reopen with the file and line — the issue body was the unfilled template, so there was nothing further to go on.

closed 2026-10-13 by mguthaus