VLSIDA/OpenRAM #228: Error in SRAM creation with banks

KanishR1, 2024-02-20:
In OpenRAM when creating an SRAM with the following configuration

```word_size = 16
num_words = 256
words_per_row = 1
tech_name = "sky130"
num_banks = 2		
num_rw_ports = 1
num_r_ports = 0
num_w_ports = 0
num_spare_rows = 1
num_spare_cols = 1
process_corners = [ "TT" ]
supply_voltages = [ 3.3 ]
temperatures = [ 25 ]
output_path = "sram_32x32"
output_name = "sram_32x32"
num_threads = 25
num_sim_threads = 25
verbose=10
drc_name = "magic"
lvs_name = "netgen"
pex_name = "magic"
```
It throws error:
```
ERROR: file simulation.py: line 605: Could not find bl net in timing paths.
Traceback (most recent call last):
  File "sram_compiler.py", line 76, in <module>
    s.save()
  File "/home/iiitb/tools/OpenRAM/compiler/sram.py", line 130, in save
    d.analysis_init(probe_address, probe_data)
  File "/home/iiitb/tools/OpenRAM/compiler/characterizer/delay.py", line 1288, in analysis_init
    self.set_internal_spice_names()
  File "/home/iiitb/tools/OpenRAM/compiler/characterizer/simulation.py", line 520, in set_internal_spice_names
    bl_name_port, br_name_port = self.get_bl_name(self.graph.all_paths, port)
  File "/home/iiitb/tools/OpenRAM/compiler/characterizer/simulation.py", line 624, in get_bl_name
    bl_names.append(self.get_alias_in_path(paths, int_net, cell_mod, exclude_set))
  File "/home/iiitb/tools/OpenRAM/compiler/characterizer/simulation.py", line 605, in get_alias_in_path
    debug.error("Could not find {} net in timing paths.".format(internal_net), 1)
  File "/home/iiitb/tools/OpenRAM/compiler/debug.py", line 48, in error
    assert return_value == 0
AssertionError
```

When the num_banks is set to 1, the SRAM memory is created. Is the memory bank functionality working in OpenRAM

cannotgetaname, 2024-02-29:
me too
I run in freepdk45 with 2 banks.

c-93, 2024-03-27:
I am getting a similar error without touching the bank configuration, by just increasing the memory size of the single port tiny example to an arbitrary (?) size. Here is an example config, which fails for me.

```python
word_size = 32 # Bits
num_words = 256
human_byte_size = "{:.0f}kbytes".format((word_size * num_words)/1024/8)

# Allow byte writes
write_size = 8 # Bits # does not matter

# Single port
num_rw_ports = 1 #one rw port will provoque the error 
num_r_ports = 0
num_w_ports = 0
num_spare_rows = 1
num_spare_cols = 1
ports_human = '1rw'

import os
exec(open(os.path.join(os.path.dirname(__file__), 'sky130_sram_common.py')).read())
```

Tail, of a massive 256mb log file:
```log
[2024-03-22 14:44:26.110332] [openram.verify.magic/run_lvs]: Instance: sky130_sram_1kbyte_1rw_32x256_8_bank_0//sky130_sram_1kbyte_1rw_32x256_8_sky130_capped_replica_bitcell_array_0// |Instance: sky130_sram_1kbyte_1rw_32x256_8_bank:bank0/sky130_sram_1kbyte_1rw_32x256_8_sky130_capped_replica_bitcell_array: 
[2024-03-22 14:44:26.110526] [openram.verify.magic/run_lvs]:   (1,3) = (52787,2)                                                                                                       |  (1,3) = (52528,2)                                                                                                       
[2024-03-22 14:44:26.110719] [openram.verify.magic/run_lvs]:   2 = 2                                                                                                                   |  2 = 2                                                                                                                   
[2024-03-22 14:44:26.110916] [openram.verify.magic/run_lvs]:   4 = 52787                                                                                                               |  4 = 52528                                                                                                               
[2024-03-22 14:44:26.111112] [openram.verify.magic/run_lvs]:                                                                                                                           |                                                                                                                          
[2024-03-22 14:44:26.111305] [openram.verify.magic/run_lvs]: Instance: sky130_sram_1kbyte_1rw_32x256_8_bank_0//sky130_sram_1kbyte_1rw_32x256_8_sky130_capped_replica_bitcell_array_0// |Instance: sky130_sram_1kbyte_1rw_32x256_8_bank:bank0/sky130_sram_1kbyte_1rw_32x256_8_sky130_capped_replica_bitcell_array: 
[2024-03-22 14:44:26.111498] [openram.verify.magic/run_lvs]:   (1,3) = (52787,2)                                                                                                       |  (1,3) = (52528,2)                                                                                                       
[2024-03-22 14:44:26.111691] [openram.verify.magic/run_lvs]:   2 = 2                                                                                                                   |  2 = 2                                                                                                                   
[2024-03-22 14:44:26.111895] [openram.verify.magic/run_lvs]:   4 = 52787                                                                                                               |  4 = 52528                                                                                                               
[2024-03-22 14:44:26.112093] [openram.verify.magic/run_lvs]:                                                                                                                           |                                                                                                                          
[2024-03-22 14:44:26.112289] [openram.verify.magic/run_lvs]: Instance: sky130_sram_1kbyte_1rw_32x256_8_bank_0//sky130_sram_1kbyte_1rw_32x256_8_sky130_capped_replica_bitcell_array_0// |Instance: sky130_sram_1kbyte_1rw_32x256_8_bank:bank0/sky130_sram_1kbyte_1rw_32x256_8_sky130_capped_replica_bitcell_array: 
[2024-03-22 14:44:26.112484] [openram.verify.magic/run_lvs]:   (1,3) = (52787,2)                                                                                                       |  (1,3) = (52528,2)                                                                                                       
[2024-03-22 14:44:26.112677] [openram.verify.magic/run_lvs]:   2 = 2                                                                                                                   |  2 = 2                                                                                                                   
[2024-03-22 14:44:26.112874] [openram.verify.magic/run_lvs]:   4 = 52787                                                                                                               |  4 = 52528                                                                                                               
[2024-03-22 14:44:26.113116] [openram.verify.magic/run_lvs]:                                                                                                                           |                                                                                                                          
[2024-03-22 14:44:26.113338] [openram.verify.magic/run_lvs]: (no matching instance)                                                                                                    |Instance: sky130_ [... cut ...]
[2024-03-22 14:44:26.172561] SP: Writing to OpenRAM/macros2/sky130_sram_1kbyte_1rw_32x256_8/sky130_sram_1kbyte_1rw_32x256_8.sp
[2024-03-22 14:44:26.265993] [openram.characterizer.functional/__init__]: Random seed for functional simulation: 1711115066265554705
[2024-03-22 14:44:26.292235] ** Spice writing: 0.1 seconds
[2024-03-22 14:44:26.292334] DELAY: Writing stimulus...
[2024-03-22 14:44:26.543349] ERROR: file simulation.py: line 605: Could not find bl net in timing paths.
```

The process fails at verifying the netlist as signals/ networks are missing.

c-93, 2024-05-16:
No, we switched to DffRam for now.

tarunkothuri, 2024-06-18:
Is there any update on this issue?

OmBhatt18, 2024-10-23:
Is there any temporary workaround for this issue ? I just want to show (view)a demo layout .

mguthaus, 2026-10-13:
Status update: the non-power-of-2 part of this is being fixed in #321, which is in review.

The probe address used during characterization was always "all address bits set", which is only a real word when `num_words` is a power of 2. #321 probes the last valid word instead. I reproduced the original failure on `dev` and confirmed the fix is correct and leaves power-of-2 configurations unchanged.

If the bank-related part of this report is separate from the non-power-of-2 probe address, it would help to split it into its own issue with a config file, so it does not get closed along with #321.