VLSIDA/OpenRAM #229: Support for PROBE PDK

glzhou97, 2024-03-17:
Hi, I was wondering if could be possible to add PROBE-PDK support (fake 3nm FinFet), please. Thanks. If need significant works, we may be able to contribute but we need specific instruction on how to do it, and if PROBE provides enough technical details for OpenRAM.

https://github.com/ABKGroup/PROBE3.0

mguthaus, 2024-03-17:
I'll keep this around as it's a good idea. However, adding a new technology is a lot of work, so we would need to identify someone to do it.

glzhou97, 2024-03-17:
Could you walk me through the possible challenges for this? I may be able to identify an intern to do it.

mguthaus, 2024-03-17:
I would estimate it as a large MS project.

glzhou97, 2024-03-17:
Understood, thanks for clarifying.

mguthaus, 2026-10-16:
Closing as stale. No activity for about 2.6 years and no reproduction against current `dev`, so there is nothing actionable here as it stands.

No work has started, and nobody here has access to the PROBE PDK to do it. A port needs its technology directory contributed alongside the compiler changes or there is nothing CI can keep working — see [technology_setup.md](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/technology_setup.md). #302 (a work-in-progress ihp13sg2 port) is a useful recent reference for the shape of one.

To be clear this is closed for inactivity, not rejected. If it still affects you please reopen or file fresh, with a reproduction against current `dev` including your configuration file, technology, OS and `python3 --version`, and the temp-directory contents OpenRAM leaves on failure ([the FAQ lists which files](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself a PR against `dev` would be welcome and we will review it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md).

closed 2026-10-16 by mguthaus