VLSIDA/OpenRAM #232: ROM >= 8k fails to build

urish, 2024-03-28:
**Describe the bug**
A clear and concise description of what the bug is.

**Version**
1.2.48 (b6a6f12642df6b84facc24a77f9a6f67a0d62dab)

**To Reproduce**
Generate a 8kbyte ROM binary file, e.g.:

```shell
dd if=/dev/zero bs=1 count=8192 | tr '\000' 'U' > rom_configs/rom_8k.bin
```

Use the following configuration:
```python
word_size = 1
check_lvsdrc = True
rom_data = "rom_configs/rom_8k.bin"
data_type = "bin"
output_name = "sky130_rom_8k
output_path = "macro/{output_name}".format(**locals())

import os
exec(open(os.path.join(os.path.dirname(__file__), 'sky130_rom_common.py')).read())
```

**Expected behavior**
Should build without any errors

**Logs**
```
[openram.router.signal_escape_router/route]: Running signal escape router...
Traceback (most recent call last):
  File "/home/runner/work/sky130-rom-experiments/sky130-rom-experiments/OpenRAM/rom_compiler.py", line 65, in <module>
    r = rom()
  File "/home/runner/work/sky130-rom-experiments/sky130-rom-experiments/OpenRAM/compiler/rom.py", line 54, in __init__
    self.r.create_layout()
  File "/home/runner/work/sky130-rom-experiments/sky130-rom-experiments/OpenRAM/compiler/modules/rom_bank.py", line [86](https://github.com/TinyTapeout/sky130-rom-experiments/actions/runs/8441697713/job/23121431113#step:4:87), in create_layout
    self.route_layout()
  File "/home/runner/work/sky130-rom-experiments/sky130-rom-experiments/OpenRAM/compiler/modules/rom_bank.py", line 124, in route_layout
    self.route_escape_pins(init_bbox)
  File "/home/runner/work/sky130-rom-experiments/sky130-rom-experiments/OpenRAM/compiler/modules/rom_bank.py", line 534, in route_escape_pins
    rtr.route(pins_to_route)
  File "/home/runner/work/sky130-rom-experiments/sky130-rom-experiments/OpenRAM/compiler/router/signal_escape_router.py", line 73, in route
    self.new_pins[source.name] = new_wires[-1]
IndexError: list index out of range
make[1]: *** [Makefile:[90](https://github.com/TinyTapeout/sky130-rom-experiments/actions/runs/8441697713/job/23121431113#step:4:91): sky130_rom_8192.ok] Error 1
make: *** [Makefile:95: sky130_rom_8192] Error 2
```

(You can see the complete logs [here](https://github.com/TinyTapeout/sky130-rom-experiments/actions/runs/8441697713/job/23121431113#step:4:26))

**Additional context**
Also happens on larger sizes (tried 12k, 16k, 24k, 32k), and even with smaller sizes if we increase the value of `words_per_row`, e.g., for a 128 bytes ROM:

```shell
dd if=/dev/zero bs=1 count=128 | tr '\000' 'U' > rom_configs/rom_128.bin
```

```python
word_size = 1
words_per_row = 32

check_lvsdrc = True

rom_data = "rom_configs/rom_128.bin"
data_type = "bin"

output_name = "rom_128"
output_path = "macro/{output_name}".format(**locals())

import os
exec(open(os.path.join(os.path.dirname(__file__), 'sky130_rom_common.py')).read())
```

mguthaus, 2024-03-31:
A 1-bit ROM seems very odd and I'm not surprised that doesn't work.  Why would you only need 1-bit output? 

It would probably be better to have a 8 or even 32-bit output and then select which bit of the output is needed.

urish, 2024-03-31:
Thanks for responding! `word_size = 1` generates a 8-bit ROM. It's the same word size as defined in the 1K example ROM:

https://github.com/VLSIDA/OpenRAM/blob/b6a6f12642df6b84facc24a77f9a6f67a0d62dab/macros/rom_configs/sky130_rom_1kbyte.py#L9

mguthaus, 2024-03-31:
I see. This is actually a mistake as it should be in bits, not bytes like the SRAMs:

https://github.com/VLSIDA/OpenRAM/blob/b6a6f12642df6b84facc24a77f9a6f67a0d62dab/compiler/rom_config.py#L21

mguthaus, 2026-10-13:
Still open. #307 is a related ROM build failure that is also open, so they may share a cause.

ROM generation goes through `compiler/modules/rom_bank.py` and `compiler/rom_config.py`. Note also #273: on `stable`, `rom_bank.py` imports `lef_rom_interconnect` which is not defined for every technology, so a default `stable` clone can fail at import before it even gets to building — worth ruling that out first if you retest.

If you would like to take it, a PR against `dev` would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch, and the regression suite is run with `cd compiler/tests && make -k -j<N>`.

mguthaus, 2026-10-29:
The ROM compiler is not supported. Tracking that in #339, which also records the current state of the ROM tests and why they are excluded from the regression suite.

Closing this against #339 rather than leaving it open as a specific bug — the 8k failure is a symptom of the subsystem not being functional, not a separate defect. Thanks for the clear reproducer with the `dd` command; if ROM support is picked up again that is a useful starting case.

closed 2026-10-29 by mguthaus