VLSIDA/OpenRAM #235: Importing a custom GDS file.

YWJ226, 2024-04-13:
**Describe the bug**
Hello, I created a custom layout file in Virtuoso, but I encountered the following issue when trying to import it into OpenRAM. Could it be that OpenRAM does not support Virtuoso's GDS files? (I was able to import GDS files created from KLayout into OpenRAM before.)

  File "/home/ywj/OpenRAM/compiler/gdsMill/gdsMill/gds2reader.py", line 184, in readBoundary
    idBits = record[0:2]
TypeError: 'NoneType' object is not subscriptable

**Version**
1.2.48

mguthaus, 2024-04-13:
GDS is a standard file format and OpenRAM can read the files. I'm not sure what is going on here due to lack of info. 

What technology are you using? 
What layers are in the layout? 
Do these match the technology?

YWJ226, 2024-04-13:
The technology library I'm using is Freepdk45nm.
I only utilized the following layers.
[image]
All designs have been subjected to DRC and LVS checks。When exported from Cadence, there were 0 errors and warnings.

mguthaus, 2024-04-13:
When you export to GDS, there is a layer map file. This determines what layers in cadence get mapped to what in the GDS. What did you use for that?

YWJ226, 2024-04-13:
When I export the GDS file, I am using the layers.map file under the technology/freepdk45 directory in OpenRAM.

mguthaus, 2024-04-13:
I need more info to replicate the issue. Otherwise, I've guessed all I can.

YWJ226, 2024-04-13:
OK,after I finish drawing the layout, I export the GDS file using these options below.
[image]

YWJ226, 2024-04-13:
I'm sorry to bother you again. What other information do I need to provide?

closed 2024-04-26 by YWJ226