VLSIDA/OpenRAM #236: Routing error due to high (vertical) density?

c-93, 2024-04-15:
Thanks for this great project.

**Describe the bug**
SupplyRouter fails to route some signals (typically VDD) for certain memory configurations.

```bash
[openram.router.supply_router/route]: Running router for vdd and gnd...
ERROR: file supply_router.py: line 84: Couldn't route from (vdd layer=m3 ll=v[191.795,-55.435] ur=v[192.285,-54.945]) to (vdd layer=m3 ll=v[191.635,-44.165] ur=v[192.125,-43.675]).
```

**Version**
1.2.48, [b6a6f12](https://github.com/VLSIDA/OpenRAM/commit/b6a6f12642df6b84facc24a77f9a6f67a0d62dab)

**To Reproduce**
This configuration should reproduce the bug:

```python
word_size = 64 # Bits
num_words = 256
words_per_row = 1 # Con also be reproduced with other dimensions
human_byte_size = "{:.0f}kbytes".format((word_size * num_words)/1024/8)

# Allow byte writes
write_size = word_size # Bits

# Dual port
num_rw_ports = 1
num_r_ports = 0
num_w_ports = 0
ports_human = '1rw'

num_spare_cols = 5 # Can also be reproduced with smaller numbers
num_spare_rows = 5 # Can also be reproduced with smaller numbers

import os
exec(open(os.path.join(os.path.dirname(__file__), 'sky130_sram_common.py')).read())
```

**Expected behavior**
Don't crash. Don't miss any signals / nets. If routing is impossible, increase (vertical) area to ease routing, or move pins more to the left. Optionally, place pins on the west and east sides (not only south) to gain even more interface surface area to further ease routing.

**Screenshots**
The individually unroutable signal depents on the memory configuration. For the provided example, the error can be located in the generated gds files.
temp.gds:
[image]

error.gds:
[image]

zoomed-in error.gds:
[image]

**Logs**
As the screenshots have a limited resolution, you may use the actual GDS files to further zoom in.
[gds.tar.gz](https://github.com/VLSIDA/OpenRAM/files/14981789/gds.tar.gz)

Log:
[sky130_sram_custom.log](https://github.com/VLSIDA/OpenRAM/files/14981797/sky130_sram_custom.log)

**Additional context**
I created 120 different SKY130 memory macro configurations and executed them on our slurm cluster for 3 days.
I swept over different port configurations (rw, rw+r), word width(1, 4, 8, 16, 32, 64 byte), and memory sizes (64 bit... 128 kByte). 

Openram finished only 10 macros. The other 110 configurations failed or ran into the SLURM job timeout after 96h.  
While some configurations quickly failed for obvious reasons (the desired total size or word width was too small or too large), others failed due to internal problems (or other user error).

I can share all 120 configurations, and provide corresponding results (1+ gigabyte tar.gz file).

**Questions:**
Is there an easy fix to prevent a design from unroutability? E.g. provide more layers, give more vertical space?

mguthaus, 2024-04-15:
Thanks for the suggested improvements. We likely won't get to improving the supply router for a while. We are constrained by resources right now.

mguthaus, 2026-10-13:
Closing as stale. This has been inactive for 2.5 years and there is no current reproduction against `dev`, so it is not actionable as it stands.

The router has changed since, including the supply-router and bbox-tree fixes merged in #314 and #317.

To be clear, this is being closed for inactivity rather than rejected. If it still affects you, please reopen or file a fresh issue with a reproduction against current `dev`, including your configuration file, technology, OS and `python3 --version`, and the contents of the temp directory OpenRAM leaves behind on failure ([the FAQ explains which files those are](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself, a PR would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch.

closed 2026-10-13 by mguthaus