VLSIDA/OpenRAM #238: Import custom file

YWJ226, 2024-04-26:
**Describe the bug**
Hello, I found that when importing my custom sense amplifier into OpenRAM, it overlaps two sense amplifiers together. It defines the width of my custom sense amplifier as the width of the built-in sense amplifier in OpenRAM. I'm wondering how to solve this issue.The width of my custom sense amplifier is shown in Figure 1.Figure 2 depicts the width of the sense amplifier generated by OpenRAM for the SRAM.

[image]

[image]

**Version**
1.2.48

**To Reproduce**
I only added this line of code in `tech.py`, and I added the corresponding `.gds` and `.sp` files.
[image]

**Expected behavior**
I hope to prevent the generated sense amplifier layout from overlapping anymore.

mguthaus, 2024-04-26:
OpenRAM uses two things for the width and height:
1) If you specify a boundary shape, it will use this. This can have shapes extending outside of it, etc.
2) If no boundary is provided, it uses the bounding box of all the shapes.

closed 2024-04-26 by mguthaus

YWJ226, 2024-04-27:
Thank you for your guidance. May I ask how I should specify the boundary shape? Should I specify the boundary shape when drawing the layout, or do I need to modify a parameter within the OpenRAM code?

YWJ226, 2024-04-27:
Hello, I'm sorry to bother you again. I have encountered another issue now. I have read your article from 2013: "OpenRAM: An Open-Source Memory Compiler," which mentions, "The two major considerations while designing the sense amplifier cell are the size of the cell and the bit line/input pitches." Could you please advise on how to align the bit line inputs of the sense amplifier with the outputs of the write driver? I have already adjusted the pitch of BL to be the same, but I still encounter metal2 connection errors, as shown in the diagram below.
[image]